WildernessLabs/Meadow_Issues #2: `Task`s execute strangely

bryancostanich, 2019-03-20:
System.Thread works now in the new new build, but Tasks are acting strangely. If you run the [Basic_Threading](https://github.com/WildernessLabs/Meadow.Core/blob/master/source/Tests/Basic_Threading/ThreadingApp.cs) app, you'll see that the following code, works as expected:

```csharp
public class ThreadingApp : AppBase<F7Micro, ThreadingApp>
{
    IDigitalOutputPort out1;
    IDigitalOutputPort out2;

    public ThreadingApp()
    {
        out1 = Device.CreateDigitalOutputPort(Device.Pins.D00);
        out2 = Device.CreateDigitalOutputPort(Device.Pins.D01);

        out1.State = true;

        StartAThread();

        Thread.Sleep(Timeout.Infinite);
    }

    public void StartAThread()
    {
        Thread th = new Thread(() => {
            while (true) {
                out2.State = true;
                Thread.Sleep(250);
                out2.State = false;
                Thread.Sleep(250);
            }
        });
        th.Start();
    }
}
```

That is, an LED attached to `D01` will be on, while an LED attached to `D01` will blink on and off in 250ms increments.

However, if we convert that code to Tasks, as seen in the [Basic_Tasks](https://github.com/WildernessLabs/Meadow.Core/tree/master/source/Tests/Basic_Tasks) app:

```csharp
public class TasksApp : AppBase<F7Micro, TasksApp>
{
    IDigitalOutputPort out1;
    IDigitalOutputPort out2;

    public TasksApp()
    {
        out1 = Device.CreateDigitalOutputPort(Device.Pins.D00);
        out2 = Device.CreateDigitalOutputPort(Device.Pins.D01);

        out1.State = true;

        StartATask();
    }

    public void StartATask()
    {
        Task t = new Task(() => {
            while (true) {
                out2.State = true;
                Task.Delay(250);
                out2.State = false;
                Task.Delay(250);
            }
        });
        t.Start();
    }
}
```

Then an LED on `D00` will go on, as expected, but the LED at `D01` will flicker during its `ON` state. It's almost as if the `Task` itself is somehow being timesliced. It's very strange.

rdavisau, 2019-05-21:
I think the delays be `await`ed.

```
    public void StartATask()
    {
        Task t = new Task(async () => {
            while (true) {
                out2.State = true;
                await Task.Delay(250);
                out2.State = false;
                await Task.Delay(250);
            }
        });
        t.Start();
    }
```

without that, the loop is effectively doing the below, which would explain flickering:

```
            while (true) {
                out2.State = true;
                out2.State = false;
            }
```

tritao, 2019-05-28:
@rdavisau Thanks for the feedback, makes sense.

We'll close this as soon as we test that it works with the await.

bryancostanich, 2019-06-09:
@rdavisau you're correct! thanks for the fix. :)

closed 2019-06-09 by bryancostanich