WildernessLabs/Meadow_Issues #3: `Debug.WriteLine` doesn't output, but `Console.WriteLine` does.

bryancostanich, 2019-03-20:
If you do the following in a Meadow app, it works as expected; you get console output to the semihosted stlink window:

```csharp
Console.WriteLine("Hello, World.");
```

However, if you use `System.Diagnostics.Debug`, no output is seen:

```csharp
Debug.WriteLine("Hello, from debug.");
```

LordBenjamin, 2019-11-21:
As per #41, this doesn't just fail to output for me - it actually crashes:

```
[0x4:] EXCEPTION handling: System.IO.FileNotFoundException: Could not load file or assembly 'System.Configuration, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies.
```

KiwiBryn, 2020-03-01:
+∞ Just lost hours from my life due to Debug.WriteLine buried deep in 3rd party library

bryancostanich, 2021-05-10:
Ok, so the good news here is that this no longer crashes, but the output never makes it to the CLI. 

@Peter-Moody assigning this to you, lower priority. You might need to synch with @alexischr to figure out how to map this to stdout or whatever.

bryancostanich, 2022-07-03:
Has anyone tested this recently?

Peter-Moody, 2022-07-13:
IIRC Debug.Writeline only exists in a debug build. Are our binaries always built as debug?
From https://docs.microsoft.com/en-us/dotnet/api/system.diagnostics.tracelistener?view=net-6.0
"To enable debugging in C#, add the /d:DEBUG flag to the compiler command line when you compile your code, or you can add #define DEBUG to the top of your file."

zivkan, 2022-07-15:
> Are our binaries always built as debug?

In Visual Studio, the build "configuration" is a drop down, by default at the top of the window, in the toolbar. It's been a while since I tried meadow development, but if it's possible to build with the `dotnet` cli, then it defaults to Debug, and you can change it `dotnet build -c Release`. Similarly for msbuild `msbuild /p:Configuration=Release`. These are defaults for all .NET development, not specific to Meadow.

bryancostanich, 2023-10-17:
On Alexis's plate after the OtA bug. Should be an easy bug.

ctacke, 2023-10-24:
Here's a stack trace for the failure
```
System.Configuration.ConfigurationErrorsException: Error Initializing the configuration system.
at System.Configuration.ClientConfigurationSystem.get_Configuration () [0x0002a] in <65bb65e201494ea1a4c08c8b727f1bf7>:0
at System.Configuration.ClientConfigurationSystem.System.Configuration.Internal.IInternalConfigSystem.GetSection (System.String configKey) [0x00000] in <65bb65e201494ea1a4c08c8b727f1bf7>:0
at System.Configuration.ConfigurationManager.GetSection (System.S
ring sectionName) [0x00005] in <65bb65e201494ea1a4c08c8b727f1bf7>:0
at System.Configuration.PrivilegedConfigurationManager.GetSection (System.String sectionName) [0x00000] in <7d08243dabbe444c8087ce5162c8810a>:0
at System.Diagnostics.DiagnosticsConfiguration.GetConfigSection () [0x00000] in <7d08243dabbe444c8087ce5162c8810a>:0
at System.Diagnostics.DiagnosticsConfiguratio
.Initialize () [0x0002a] in <7d08243dabbe444c8087ce5162c8810a>:0
at System.Diagnostics.DiagnosticsConfiguration.get_IndentSize () [0x00000] in <7d08243dabbe444c8087ce5162c8810a>:0
at System.Diagnostics.TraceInternal.InitializeSettings () [0x0004e] in <7d08243dabbe444c8087ce5162c8810a>:0
at System.Diagnostics.TraceInternal.get_UseGlobalLock () [0x00000] in <7d08243dabbe444
8087ce5162c8810a>:0
at System.Diagnostics.TraceInternal.WriteLine (System.String message) [0x00000] in <7d08243dabbe444c8087ce5162c8810a>:0
at System.Diagnostics.Debug.WriteLine (System.String message) [0x00000] in <7d08243dabbe444c8087ce5162c8810a>:0
```

duduita, 2023-12-20:
I was getting this exception also:
`System.MissingMethodException: Default constructor not found for type System.Configuration.ExeConfigurationHost.`

I realized that it was happening when the `CreateInstance` method was being called:
`host = (IInternalConfigHost) Activator.CreateInstance (typeConfigHost);`

Then, after reading [this](https://stackoverflow.com/a/8950365), I tried to skip linking (trimming) the `System.Configuration` (`meadow app deploy -n System.Configuration`), since it could be removing some necessary types. 

After doing that, I'm not getting more exceptions, although it should not be a definite solution, it indicates that the exception was caused by linking issues in fact.

alexischr, 2024-01-09:
Fixed in 1.7

closed 2024-01-09 by alexischr