WildernessLabs/Meadow_Issues #4: Debug.WriteLine crashes App in an Ubuntu 12.04 Virtual Machine in Windows

iruel865, 2019-03-22:
Trying to run any Meadow.Core sample project that has a Debug.WriteLine instruction makes the app throw an exception:

[1:] EXCEPTION handling: System.IO.FileNotFoundException: Could not load file or assembly 'System.Configuration, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies.

Unhandled Exception:
System.IO.FileNotFoundException: Could not load file or assembly 'System.Configuration, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies.
File name: 'System.Configuration, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'
  at System.Diagnostics.Debug.WriteLine (System.String message) <0xc01dd600 + 0x00008> in <23340a11bb41423aa895298bf881ed68>:0
  at HelloLED.LEDApp.CreateOutputs () <0xc01dd490 + 0x00008> in <2edfc67474ee45a9bfedef4018dc6769>:0
  at HelloLED.LEDApp..ctor () <0xc00ffc78 + 0x00018> in <2edfc67474ee45a9bfedef4018dc6769>:0
  at HelloLED.MainClass.Main (System.String[] args) <0xc00ffa10 + 0x00004> in <2edfc67474ee45a9bfedef4018dc6769>:0
[ERROR] FATAL UNHANDLED EXCEPTION: System.IO.FileNotFoundException: Could not load file or assembly 'System.Configuration, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies.
File name: 'System.Configuration, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'
  at System.Diagnostics.Debug.WriteLine (System.String message) <0xc01dd600 + 0x00008> in <23340a11bb41423aa895298bf881ed68>:0
  at HelloLED.LEDApp.CreateOutputs () <0xc01dd490 + 0x00008> in <2edfc67474ee45a9bfedef4018dc6769>:0
  at HelloLED.LEDApp..ctor () <0xc00ffc78 + 0x00018> in <2edfc67474ee45a9bfedef4018dc6769>:0
  at HelloLED.MainClass.Main (System.String[] args) <0xc00ffa10 + 0x00004> in <2edfc67474ee45a9bfedef4018dc6769>:0

bryancostanich, 2019-03-24:
See Issue #3. Not sure why it crashes, though.

bryancostanich, 2019-11-22:
Duplicate

closed 2019-11-22 by bryancostanich