WildernessLabs/Meadow_Issues #5: Calling DateTime.Now causes exception

bryancostanich, 2019-03-24:
If you make a call to `DateTime.Now` you'll get the following exception:

```
[3:] EXCEPTION handling: System.DllNotFoundException: libc.dylib
EXCEPTION: catch found at clause 0 of System.TimeZoneInfo:readlink (string)
[3:] EXCEPTION handling: System.IO.DirectoryNotFoundException: Could not find a part of the path "/etc/localtime".
EXCEPTION: catch found at clause 0 of System.TimeZoneInfo:FindSystemTimeZoneByFileName (string,string)
[3:] EXCEPTION handling: System.TimeZoneNotFoundException: Couldn't read time zone file /etc/localtime
EXCEPTION: catch found at clause 1 of System.TimeZoneInfo:CreateLocal ()
[3:] EXCEPTION handling: System.IO.DirectoryNotFoundException: Could not find a part of the path "/usr/share/zoneinfo/localtime".
EXCEPTION: catch found at clause 0 of System.TimeZoneInfo:FindSystemTimeZoneByFileName (string,string)
[3:] EXCEPTION handling: System.TimeZoneNotFoundException: Couldn't read time zone file /usr/share/zoneinfo/localtime
EXCEPTION: catch found at clause 1 of System.TimeZoneInfo:CreateLocal ()
```

We need to assume a time zone and such for now. let's discuss at weekly engineering.

tritao, 2019-03-25:
I've seen this one before, I have a fix pending, but didn't seem to fully work. 
Will need to revisit.

bryancostanich, 2020-02-01:
Think this got fixed, haven't seen the error in a while.

closed 2020-02-17 by bryancostanich