WildernessLabs/Meadow_Issues #10: Port the Meadow Samples to new SDK

NevynUK, 2019-09-11:
Port the samples in the [Meadow_Samples](https://github.com/WildernessLabs/Meadow_Samples) repository to the new Meadow SDK format (i.e. File -> New Meadow format.

bryancostanich, 2019-09-14:
New project format looks like the following:

```xml
<Project Sdk="Meadow.Sdk/1.0">
  <PropertyGroup>
    <TargetFramework>net472</TargetFramework>
    <OutputType>Exe</OutputType>
    <AssemblyName>App</AssemblyName>
  </PropertyGroup>
  <ItemGroup>
    <PackageReference Include="WildernessLabs.Meadow.Assemblies" Version="0.1.0" />
  </ItemGroup>
</Project>
```

Old school project style (which needs to be updated):

```xml
<?xml version="1.0" encoding="utf-8"?>
<Project DefaultTargets="Build" ToolsVersion="4.0" xmlns="http://schemas.microsoft.com/developer/msbuild/2003">
  <PropertyGroup>
    <Configuration Condition=" '$(Configuration)' == '' ">Debug</Configuration>
    <Platform Condition=" '$(Platform)' == '' ">AnyCPU</Platform>
    <ProjectGuid>{1EBFD0E5-F32C-46CD-9A60-0143EB8B8FA3}</ProjectGuid>
    <OutputType>Exe</OutputType>
    <RootNamespace>GpioInterrogation</RootNamespace>
    <AssemblyName>App</AssemblyName>
    <TargetFrameworkVersion>v4.7.2</TargetFrameworkVersion>
    <TargetFrameworkProfile />
  </PropertyGroup>
  <PropertyGroup Condition=" '$(Configuration)|$(Platform)' == 'Debug|AnyCPU' ">
    <DebugSymbols>true</DebugSymbols>
    <DebugType>full</DebugType>
    <Optimize>false</Optimize>
    <OutputPath>bin\Debug</OutputPath>
    <DefineConstants>DEBUG;</DefineConstants>
    <ErrorReport>prompt</ErrorReport>
    <WarningLevel>4</WarningLevel>
    <ExternalConsole>true</ExternalConsole>
  </PropertyGroup>
  <PropertyGroup Condition=" '$(Configuration)|$(Platform)' == 'Release|AnyCPU' ">
    <Optimize>true</Optimize>
    <OutputPath>bin\Release</OutputPath>
    <ErrorReport>prompt</ErrorReport>
    <WarningLevel>4</WarningLevel>
    <ExternalConsole>true</ExternalConsole>
  </PropertyGroup>
  <ItemGroup>
    <Reference Include="System" />
  </ItemGroup>
  <ItemGroup>
    <Compile Include="Program.cs" />
    <Compile Include="Properties\AssemblyInfo.cs" />
    <Compile Include="GpioApp.cs" />
  </ItemGroup>
  <ItemGroup>
    <ProjectReference Include="..\..\Meadow.Core\Meadow.Core.csproj">
      <Project>{B13C7BDD-77F5-45F8-BDB9-6CD368C67410}</Project>
      <Name>Meadow.Core</Name>
    </ProjectReference>
  </ItemGroup>
  <Import Project="$(MSBuildBinPath)\Microsoft.CSharp.targets" />
</Project>
```

bryancostanich, 2019-09-14:
Note, I just ported one of these; once the project file is updated, the `/Properties` folder will need to be deleted as well.

bryancostanich, 2019-09-14:
`GpioInterrogation` sample has been updated as an example.

lamebrain, 2019-09-14:
It should be <Project Sdk="Meadow.Sdk"> so that it picks up the latest sdk version

bryancostanich, 2019-09-14:
Heh. will fix.

bryancostanich, 2019-09-14:
Needs to be `<Project Sdk="Meadow.Sdk/1">` for now.

bryancostanich, 2019-09-14:
Thanks, @Patridge !

closed 2019-09-14 by bryancostanich