WildernessLabs/Meadow_Issues #13: Memory leak

IngenuityMicro, 2019-10-01:
The following SPI code throws a memory allocation error after ~192k iterations

+SendData 0A-0A-0A-00-0A-0A-0A-0A-0A-0A-00-00-0A-0A-00-0A-00-00-00-00-00-00-00-00-00-00-00-0A-0A-00-0A-0A-00-0A-00-00
Could not allocate 300 (1 * 300) bytes

public MeadowApp()
        {
            ConfigurePorts();
            greenLed.State = true;
            cs.State = true;
            _data = new byte[36];
            for (; ; )
            {
                for(int i =0; i < 1000; i++)
                {
                    Display(i);
                    cnt++;
                    Console.WriteLine(cnt.ToString());
                }            
            }         
        }

adrianstevens, 2019-10-29:
Thanks @IngenuityMicro - it looks like we need to improver the behaviour of the garbage collector. I'd recommend using persistent buffers for now instead of relying on the GC, especially within loops.

bryancostanich, 2020-06-05:
Fix in flight for b4.0

bryancostanich, 2020-06-18:
@IngenuityMicro can you give a full repro app code please? i think we have a fix for this, and i want to test it, but the code above is missing stuff.

bryancostanich, 2020-07-06:
@IngenuityMicro I think we've fixed this, but we really could use a full repro, there's too much missing in the code above to infer what we need to do.

closed 2020-07-15 by bryancostanich