WildernessLabs/Meadow_Issues #14: ISpiBus ExchangeData Bug

GeorgeStephens, 2019-10-04:
The current [SPI](http://beta-developer.wildernesslabs.co/docs/api/Meadow/Meadow.Hardware.IByteCommunications.html) interface resolves around using devices with register addresses no longer that one byte though some devices have register addresses that are more than one byte long. 

To work around this I have switched to using `ExchangeData` from the underlying [ISpiBus](http://beta-developer.wildernesslabs.co/docs/api/Meadow/Meadow.Hardware.ISpiBus.html) that I assume does what I need; send x bytes before receiving x bytes. I say assume because when executed the clock output is only on for as long as it is writing data and not when a response from the device is expected as per the bellow image. 

`byte[] data = new byte[2];`
`data = _spiBus.ExchangeData(_cs, new byte[2] { 0x80, 0x01 });`

[image]

If this is the intended behavior can the API be changed to for the clock to remain active for long enough to receive x bytes? Maybe extend the API to say something like:
`byte[] ExchangeData(IDigitalOutputPort chipSelect, params byte[] dataToWrite, int numberOfBytesToRecieve))`?

It might be worth noting that writing the data then reading is not an option for two reasons:
1. Between sending and receiving the Chip Select pin is toggled so the slave device takes it as two separate requests.
2. Even when using a separate pin as the chip select the delay between the commands is around 160ms, even assuming that Meadow is far, far from optimised this will certainly not come down to the 3-4uS expected on a SPI bus.

Bellow is the code sample and output image from writing both commands separately:

`_spiBus.SendData(_cs, new byte[2] { 0x80, 0x01 });`
`data = _spiBus.ReceiveData(_cs, 2);`

[image]

ctacke, 2019-10-12:
Sorry for not getting back to you on this sooner, and for not documenting ExchangeData better.  I just put some comments into the code this morning to help clarify, so that should appear in the next release, but here it is:

<Due to the nature of a data exchange on a SPI bus, equal numbers of bytes must always be transmitted and received.  The length of the **dataToWrite** parameter dictates how many bytes will be exchanged.
>The **dataToWrite** buffer will start transmitting on the first clock cycle of the exchange.  If you want the output data to start transmitting on a later clock cycle, you must left-pad **dataToWrite** with a zero for each clock cycle to skip (it is not actually skipped, but a zero will be transmitted on those cycles).
>The data buffer returned from ExchangeData will be the same size as the **dataToWrite** parameter.  If you want to read more data that you are writing, you must right-pad the input parameter with enough empty bytes (zeros) to account for the desired return data.

ctacke, 2019-10-12:
To be clear, ExchangeData is used for when you need full duplex transmission (i.e. data going both ways at the same time).  If you want to write and then receive, you can use SendData followed by ReceiveData (though, because of the way SPI actually works, it calls ExchangeData internally anyway).

closed 2019-10-12 by ctacke

GeorgeStephens, 2019-10-12:
And the realization of the fact I could pad it out two further blank bytes came just as I read 

>it calls ExchangeData internally anyway

thus based on my original code I will send `byte[] data = _spiBus.ExchangeData(_cs, new byte[4] { 0x80, 0x01, 0x00, 0x00 });` and the last two bytes will contain my response! 

Thanks Chris