WildernessLabs/Meadow_Issues #16: DFU-Util finds no devices

nychold, 2019-10-14:
So I got my Meadow a few days ago, and finally had the change to fool around with it tonight.  I downloaded DFU-Util and was on my way to installing VS2019 when I tested dfu-util --list.  Nothing comes up.  I have heard of problems before with microcontrollers and USB 3 ports (which all of mine are), so I tested it with a USB 2.0 hub which works in most cases.

But so far, nothing.  When I boot the board, it starts with a blinking LED going through a 7 color spectrum (cool, actually), but when I hold BOOT, then reset the board, I don't get any indicators at all.

I tried Zadig, but Zadig doesn't see any devices installed at all.  At this point, I don't know what to do.  I'm running Windows 10 (Version 10.0.17763 Build 17763).  Any help would be appreciated.  Thanks.

gismofx, 2019-10-15:
Try a another usb cord...?

nychold, 2019-10-15:
Well, that was embarrassing.  The same cable I've used for years with FPGAs doesn't work anymore.  Tried a different cable, worked fine.  Oh well, lesson learned. :)

closed 2019-10-15 by nychold