WildernessLabs/Meadow_Issues #18: PushButton only works with constructor where Device is passed

padesso, 2019-10-25:
**Summary**
When creating a Meadow.Foundation.Sensors.Buttons.PushButton, the event handlers only fire when the constructor used is where the IIODevice is passed as a parameter.

**How To Reproduce**
Instantiate a PushButton as follows:
PushButton buttonPort = new PushButton(Device.CreateDigitalInputPort(Device.Pins.D04));

Add event handlers:
buttonPort.PressStarted += ButtonPort_PressStarted;
buttonPort.PressEnded += ButtonPort_PressEnded;

**Workaround**
To get the event handlers to fire, the alternate constructor override may be used:
PushButton buttonPort = new PushButton(Device, Device.Pins.D04, 100);

Add event handlers:
buttonPort.PressStarted += ButtonPort_PressStarted;
buttonPort.PressEnded += ButtonPort_PressEnded;

adrianstevens, 2019-10-29:
@jorgedevs - this should be an easy fix, let's take a look this week

bryancostanich, 2019-12-16:
@ctacke did this get fixed in b3.6 with the other pushbutton fixes?

ctacke, 2019-12-29:
The issue here is that the `IPort` used by the `PushButton` must have interrupts enabled.  If you pass in just an IPin, the PushButton creates the Port for you with interrupts (rising and falling) enabled.  I've updated the MF code to throw a meaningful exception if you try wiring up a PushButton event without the appropriate interrupt enabled on the underlying port. This will be in Beta3.7

closed 2020-02-17 by bryancostanich