WildernessLabs/Meadow_Issues #21: Issues with multiple buttons

mtboeke, 2019-10-29:
I get wonky behavior when trying to use multiple buttons.

As a test, I added a PushButton on all 16 digital pins.  When I did that, several pins (specifically D00, D02, D04, D05, D14) did not fire events.

Even with a smaller test of just pins D00-D04, D00 would not work when D04 was enabled, but it would work individually, or if only 4 buttons were attached.  My code for this test is attached.

[image]

bryancostanich, 2019-11-20:
fixing/work item for Beta3.6

bryancostanich, 2020-02-01:
Fixicating for `b3.7`.

ctacke, 2020-02-01:
This specific issue is fixed, but the test above still has problems due to [another active bug](https://github.com/WildernessLabs/Meadow_Issues/issues/59)

closed 2020-02-17 by bryancostanich