WildernessLabs/Meadow_Issues #22: I2C in SSD1306 is slooooooow

mtboeke, 2019-10-29:
Using the SSD1306 class, my display takes ~8 seconds to clear and refresh.  Sample code attached.

[image]

bryancostanich, 2019-11-20:
App code is interpreted right now. We're working AOT support for Beta4.

SupraBitKid, 2019-11-28:
I believe there is something more going on.  Interpreted code doesn't look like the only problem.

This code always displays zero:
`			var busForDisplay = Device.CreateI2cBus( 1000 ) as I2cBus;`
`			Console.WriteLine( "Frequency: {0}", busForDisplay.Frequency );`

SupraBitKid, 2019-12-01:
The following kludge allows me to speed up the refresh:

`namespace Meadow.Hardware {

	public static class I2cBusExtensions {

		public static void SetFrequency( this I2cBus device, uint frequency ) {

			var fields = typeof( I2cBus ).GetFields( System.Reflection.BindingFlags.Instance
						   | System.Reflection.BindingFlags.GetField
						   | System.Reflection.BindingFlags.SetField
						   | System.Reflection.BindingFlags.NonPublic );

			for( int index = 0; index < fields.Length; index++ )
				if( fields[ index ].Name.Contains( "Frequency" ) ) {
					fields[ index ].SetValue( device, frequency );
					return;
				}
		}
	}
}`

bryancostanich, 2020-02-01:
@ctacke looks like the frequency property might be buggered.

mtboeke, 2020-02-18:
Tested this in 0.3.7 and display refresh is down to ~1500ms.  Still too slow for real use, but a vast improvement!

MikroBusNet, 2020-02-19:
This one should not be closed as it is not fixed in Beta 3.7. 

The constructor still asks for a ushort, which cannot handle 100 or 400 KHz (100.000 or 400.000 parameter value).

MikroBusNet, 2020-02-19:
FWIW,
the workaround suggested by SupraBitKid with his extension does the job. I can at least have (almost) 100 KHz on my I2C display.
I say "almost" because the measired frequency is 95.4/96.5 KHz, not exactly 100 KHz.

I don't have any 400 KHz device yet so I can't say much more.

bryancostanich, 2020-03-09:
b3.8

closed 2020-03-09 by bryancostanich