WildernessLabs/Meadow_Issues #27: I2C Pin labels / diagrams are incorrect in documentation

NevynUK, 2019-11-02:
Documentation has I2C pins listed as D06 and D07 but the board has the pins listed as D07 and D08.

Documentation and diagrams need to be updated.

bryancostanich, 2019-11-02:
@NevynUK can you point us where? I updated the pinout diagram here: http://beta-developer.wildernesslabs.co/Meadow/Meadow_Basics/IO/

bryancostanich, 2019-11-02:
Looks like this diagram is wrong: http://beta-developer.wildernesslabs.co/Meadow/Meadow_Basics/IO/Digital/Protocols/I2C/

Any other places?

NevynUK, 2019-11-02:
The following text appears on the I2C page:

_The I2C CLK pin can be found on the Meadow F7 Micro labeled D07.

Data Signal (DAT)
The data signal wire carries the actual messages and can be found on the F7 Micro pin labeled D06._

jorgedevs, 2019-11-03:
Done! Updated I2C diagram along with the texts mentioning the wrong pins.

http://beta-developer.wildernesslabs.co/Meadow/Meadow_Basics/IO/Digital/Protocols/I2C/

closed 2019-11-03 by jorgedevs