WildernessLabs/Meadow_Issues #28: Program Stops Running after a few minutes. - Sample Code

gismofx, 2019-11-02:
The following sample code will stop running after a few minutes. If you comment out the first two rotary encoder calls, the code will not freeze up. Another twist, if you replace the the 'step' with a PWM(on the same pin) instead of a DigitalOutputPort, it will not freeze up.

```
public class MeadowApp : App<F7Micro, MeadowApp>
	{
		public MeadowApp()
		{
			var re = new RotaryEncoder(Device, Device.Pins.D10, Device.Pins.D11);
			re.Rotated += Re_Rotated;
			
			var step = Device.CreateDigitalOutputPort(Device.Pins.D03);
			var dir = Device.CreateDigitalOutputPort(Device.Pins.A05);
			//var delay = new TimeSpan(0,0,0,0,1);
			var delay = 1;
			while (true)
			{	
				for (int i = 1; i < 200; i++)
				{
					step.State = true;
					Thread.Sleep(delay);
					step.State = false;
					Thread.Sleep(delay);
				}
				Thread.Sleep(100);
				dir.State = !dir.State;
			}
	        }
		private void Re_Rotated(object sender, Meadow.Peripherals.Sensors.Rotary.RotaryTurnedEventArgs e)
		{
			Console.WriteLine("Rotated:" + e.Direction.ToString());
		}
	}
```

bryancostanich, 2019-11-22:
I believe this is related to this: https://github.com/WildernessLabs/Meadow_Issues/issues/37

Gonkers, 2020-03-07:
This is interesting to me. I am noticing the same behavior with the hello meadow sample app. It will run for a while, then just lock up. I stop getting console output and the LED stops pulsing.

<details><summary>Code</summary>
<p>

```csharp
using System;
using System.Threading;
using Meadow;
using Meadow.Devices;
using Meadow.Foundation;
using Meadow.Foundation.Leds;

namespace MeadowApplication1
{
    public class MeadowApp : App<F7Micro, MeadowApp>
    {
        const int pulseDuration = 3000;
        RgbPwmLed rgbPwmLed;

        public MeadowApp()
        {
            rgbPwmLed = new RgbPwmLed(Device,
                       Device.Pins.OnboardLedRed,
                       Device.Pins.OnboardLedGreen,
                       Device.Pins.OnboardLedBlue);

            PulseRgbPwmLed();
        }

        protected void PulseRgbPwmLed()
        {
            while (true)
            {
                Pulse(Color.Red);
                Pulse(Color.Green);
                Pulse(Color.Blue);
            }
        }

        protected void Pulse(Color color)
        {
            rgbPwmLed.StartPulse(color);
            Console.WriteLine($"Pulsing {color}");
            Thread.Sleep(pulseDuration);
            rgbPwmLed.Stop();
        }
    }
}
```

</p>
</details>

upswing1, 2020-03-17:
I have the same exact problem as Gonkers describes..  The application freezes after a certain time

Vinz68, 2020-03-22:
I have the same problem on my 2 DevBoards. The standard template (StartPulse) just stops after some time. Changed to StartBlink / same problem. Reset and Renew of file system does not solve it.

zivkan, 2020-03-28:
I'm really not a fan of "+1" or "me too" posts, but I hope my post might give some helpful information.

Using the sample app, I added a `DateTime` field to the class, initialise it with `DateTime.Now` in the constructor, then changed the `Console.WiteLine` line in `Pulse` to:

```cs
DateTime now = DateTime.UtcNow;
var duration = now - start;
Console.WriteLine($"[{duration}] Pulsing {color}");
```

The device always stops responding just after 13 minutes and 20 seconds. I have never once seen it report 13 minutes and 30 seconds. It's still happening for me with 0.3.9. I previously tested with 0.3.7.

Adding 
```cs
for (int i = 0; i < 10000; i++)
{
    Console.WriteLine("iteration = " + i);
}
```
to the constructor causes the device to freeze before finishing the loop. On my device, it always freezes around iteration 760.

I didn't yet test with 0.3.9, but with 0.3.7, even when I removed all `Console.WriteLine` statements, the device still stopped responding after some time, but I'm not patient enough to watch it carefully with a stopwatch, so I don't know if it's still after 13 minutes or if it lasted longer.

FraserElectronics, 2020-04-15:
Has this issue been fixed yet ?
I am seeing Meadow (running the default flash LEDs app) stopping consistently after 11 minutes 30 seconds. I am running Meadow OS 0.3.10.
Kind of makes the board useless for my needs at the moment.

Andy

stickybun, 2020-04-29:
For me, this seemed entirely random at first, but turned out was something to do with the GC (Garbage collector) not collecting before the device ran out of memory - or the collection itself was stopping the device, as would always hit ~4MB when it became non-responsive:

Repro:
```csharp
while(true)
{
    var mem = GC.GetAllocatedBytesForCurrentThread() / 1024f / 1024f;
    var total = GC.GetTotalMemory(forceFullCollection:false) / 1024f / 1024f;
    var bigString = String.Join(", ", Enumerable.Range(0, 1000).Select(m => "Build a large number of strings that should require a GC"));

    Console.WriteLine("{0}: GC.GetAllocatedBytesForCurrentThread: {1:F2}MB, GC.GetTotalMemory: {2:F2}", DateTime.UtcNow, mem, total);

    Thread.Sleep(1000);
}
```

Output:
```log
Mono enabled, will run app.exe
01/01/1970 00:00:16: GC.GetAllocatedBytesForCurrentThread: 0.08MB, GC.GetTotalMemory: 4.03
01/01/1970 00:00:17: GC.GetAllocatedBytesForCurrentThread: 0.36MB, GC.GetTotalMemory: 4.29
...
01/01/1970 00:00:35: GC.GetAllocatedBytesForCurrentThread: 4.27MB, GC.GetTotalMemory: 7.99
01/01/1970 00:00:36: GC.GetAllocatedBytesForCurrentThread: 4.53MB, GC.GetTotalMemory: 8.24
01/01/1970 00:00:37: GC.GetAllocatedBytesForCurrentThread: 4.79MB, GC.GetTotalMemory: 8.49
UNRESPONSIVE
```

Workaround:
```csharp
while(true)
{
    var mem = GC.GetAllocatedBytesForCurrentThread() / 1024f / 1024f;
    var total = GC.GetTotalMemory(forceFullCollection:false) / 1024f / 1024f;
    var bigString = String.Join(", ", Enumerable.Range(0, 1000).Select(m => "Build a large number of strings that should require a GC"));

    Console.WriteLine("{0}: GC.GetAllocatedBytesForCurrentThread: {1:F2}MB, GC.GetTotalMemory: {2:F2}", DateTime.UtcNow, mem, total);

    if (total > 8)
    {
        Console.WriteLine("collect\n\r");
        GC.Collect();
    }

    Thread.Sleep(1000);
}
```

Workaround output:
```logs
Mono enabled, will run app.exe
01/01/1970 00:00:13
01/01/1970 00:00:16: GC.GetAllocatedBytesForCurrentThread: 0.08MB, GC.GetTotalMemory: 4.03
01/01/1970 00:00:17: GC.GetAllocatedBytesForCurrentThread: 0.36MB, GC.GetTotalMemory: 4.29
01/01/1970 00:00:18: GC.GetAllocatedBytesForCurrentThread: 0.62MB, GC.GetTotalMemory: 4.53
...
01/01/1970 00:00:32: GC.GetAllocatedBytesForCurrentThread: 3.75MB, GC.GetTotalMemory: 7.50
01/01/1970 00:00:33: GC.GetAllocatedBytesForCurrentThread: 4.01MB, GC.GetTotalMemory: 7.75
01/01/1970 00:00:35: GC.GetAllocatedBytesForCurrentThread: 4.27MB, GC.GetTotalMemory: 7.99
01/01/1970 00:00:36: GC.GetAllocatedBytesForCurrentThread: 4.53MB, GC.GetTotalMemory: 8.24
collect

01/01/1970 00:00:37: GC.GetAllocatedBytesForCurrentThread: 4.79MB, GC.GetTotalMemory: 4.06
...
01/01/1970 00:00:56: GC.GetAllocatedBytesForCurrentThread: 8.97MB, GC.GetTotalMemory: 8.01
collect

01/01/1970 00:00:57: GC.GetAllocatedBytesForCurrentThread: 9.23MB, GC.GetTotalMemory: 4.06
...
01/01/1970 00:01:15: GC.GetAllocatedBytesForCurrentThread: 13.40MB, GC.GetTotalMemory: 8.01
collect
```

This is great, however, calling GC.Collect() makes my actual application become non-responsive immediately! But hopefully this should shed some light on a direction to investigate.

bryancostanich, 2020-06-05:
we've got it reproduced and the fix is in flight.

bryancostanich, 2020-06-27:
Ok, near as I can tell there are actually two issues here. The first is one to do with garbage collection, which we have fixed for the b4.0 release. The second one has to do with Threads not being cleaned up (or similar), i've described it in [issue #94](https://github.com/WildernessLabs/Meadow_Issues/issues/94), along with a sample repro app. 

I'm marking this one as fixed in next release and tracking the thread issue in #94.

closed 2020-07-15 by bryancostanich