WildernessLabs/Meadow_Issues #29: Pin D04 as DigitalOutputPin is not responsive when pin D03 is setup as PWM.

gismofx, 2019-11-03:
When Pin D03 is setup as PWM and D04 is setup as DigitalOutputPin, D04's state property does not work. 

Here's a code excerpt:

```
            var step = Device.CreatePwmPort(Device.Pins.D03,250,.5f);
            var dir = Device.CreateDigitalOutputPort(Device.Pins.D04); //change pin to D02 and it will work
            step.Start();
            While (true)
            {
                dir.State= !dir.State;  //or false //<--nothing happens when the State of dir is changed.
            }

```

bryancostanich, 2019-11-21:
Part of our stabilization work for b3.6

bryancostanich, 2020-02-16:
We've done some investigation here and understand what is happening and have a workaround, but no fix yet. 

# Workaround

To get around this, configure your PWMs first, then configure your other IOs.

# Cause

This is an artifact of NuttX's underlying PWM implementation, in which, when you configure a PWM, it configures the alt setting of all other pins on that timer channel. The result is that it clobbers any IO settings on those pins.

For background context, NuttX takes a build and forget approach to IOs. There is no dynamic runtime changes allowed, you basically, set them up at build time and then you're stuck with them. 

We're working on a fix around this, and will likely wind up side-stepping NuttX's PWM APIs altogether (as we have done with other IO APIs because of our very different approach).

bryancostanich, 2020-02-17:
We've got an UAF (ugly-ass-fix) for this. 

If you configure a PWM after an IO, we go back and reconfigure that IO as it was previous, before the PWM stuff clobbered it.

this may have weird state issues, but seems to work for now. better fix coming in a later release.

bryancostanich, 2020-02-17:
Looks like the PR was missing something, so this isn't making it into `b3.7`.

bryancostanich, 2020-03-28:
fixed in `b3.9`

closed 2020-03-28 by bryancostanich