WildernessLabs/Meadow_Issues #35: SPI Exception whne calling ReadRegister

KiwiBryn, 2019-11-15:
Porting my LoRa radio stack to Meadow. The Semetech SX127X devices use SPI and I get an exception when I call ReadRegister. (0x42 is the silicon version). I'm using an adafruit LoRa featherwing with D09 as CS.

  while (true)
            {
                Console.WriteLine("sx127xDevice.ReadRegister...");
                byte registerValue = sx127xDevice.ReadRegister(0x42);

                //Console.WriteLine("Register 0x{0:x2} - Value 0X{1:x2} - Bits {2}", 0x42, registerValue, Convert.ToString(registerValue, 2).PadLeft(8, '0'));

                Task.Delay(10000).Wait();
            }

Produces the exception

sx127xDevice.ReadRegister...
[0x4:] EXCEPTION handling: System.Exception: Both buffers must b
e equal size' (edited) 

If it is a problem with my code e.g. I have the CS or SPI bus speed wrong (I just took defaults for SPI config) or similar a more helpful error message would be great.

Thanks
@KiwiBryn

[MeadowApp.zip](https://github.com/WildernessLabs/Meadow_Issues/files/3849645/MeadowApp.zip)

ctacke, 2019-11-15:
Fixed.  Will be released with Beta 3.5

KiwiBryn, 2019-11-18:
Hi 

Downloaded the last OS binaries (14/11/2019) from (http://developer.wildernesslabs.co/Meadow/Release_Notes/Beta3/) and updated Meadow NuGet (V0.11.0) and still getting exception

==========================================================
.
Creating SPI NSS Port...
sx127xDevice Device...
sx127xDevice.ReadRegister...
[0x4:] EXCEPTION handling: System.Exception: Both buffers must b
e equal size

Did I miss anything for new release?

Would it be possible to put version number in zip file name so it is clear which version is being downloaded.

Thanks

@KiwiBryn

ctacke, 2019-11-18:
Let me make sure the fix was in the release branch.  In the meantime, you could work around it using the base ReceiveData or ExchangeData methods.

ctacke, 2019-11-18:
This is what it would look like (taken from the BME280 peripheral code):

```
public override byte[] ReadRegisters(byte startRegister, int readCount)
{
    // the buffer needs to be big enough for the output and response
    var buffer = new byte[readCount + 1];
    var bufferTx = new byte[readCount + 1];
    buffer[0] = startRegister;

    var rx = _spi.ReceiveData(_chipSelect, readCount + 1);

    // skip past the byte where we clocked out the register address
    var registerData = rx.Skip(1).Take(readCount).ToArray();

    return registerData;
}
```

bryancostanich, 2019-12-23:
Fixed in `b3.6`

closed 2019-12-23 by bryancostanich