WildernessLabs/Meadow_Issues #39: Issue deploying first app to Meadow

clark-mcnally-pariveda, 2019-11-20:
Just got my meadow in the mail this week and have been attempting to deploy my first test app to the board. I have followed the steps for [deploying meadow](http://developer.wildernesslabs.co/Meadow/Getting_Started/Deploying_Meadow/) to the board and have redone that step several times to check that wasn't the issue. The issue I am seeing is my Visual Studio for Mac just hangs and never gets past the first dot when attempting to deploy test app to the board. I have attached screenshot of VS and log from the Meadow output.

### Screenshot

[image]

### Meadow Logs

```
Mono is currently disabled and will not run applications.

/meadow/System.dll [0xa2a9cbca],/meadow/System.Core.dll [0x65d74925],/meadow/mscorlib.dll [0x87349f8d],/meadow/Meadow.dll [0xa6a257d1],/meadow/test.exe [0xbea9be82],/meadow/Meadow.Foundation.dll [0xd7f129c4]

File transfer header received with no errors

File 10% downloaded

File 20% downloaded

File 30% downloaded

File 40% downloaded

File 50% downloaded

File 60% downloaded

File 70% downloaded

File 80% downloaded

File 90% downloaded

File 100% downloaded

File Sent Successfully (checksums calculated = 0xFB8D82A4, received = 0xFB8D82A4)

File transfer header received with no errors

File 10% downloaded

File 20% downloaded

File 30% downloaded

File 40% downloaded

File 50% downloaded

File 60% downloaded

File 70% downloaded

File 80% downloaded

File 90% downloaded

File 100% downloaded

File Sent Successfully (checksums calculated = 0x6B5F1F01, received = 0x6B5F1F01)

File transfer header received with no errors

File 10% downloaded

File 20% downloaded

File 30% downloaded

File 40% downloaded

File 50% downloaded

File 60% downloaded

File 70% downloaded

File 80% downloaded

File 90% downloaded

File 100% downloaded

File Sent Successfully (checksums calculated = 0x4C1D20BD, received = 0x4C1D20BD)

Mono being enabled. Restarting F7 Micro

Mono is currently enabled to run applications.

.
```

clark-mcnally-pariveda, 2019-11-20:
After going back through deploy meadow steps I noticed a few errors from dfu-util when flashing Meadow to the board.

### Terminal Logs

```
dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 241882
Download	[=========================] 100%       241882 bytes
Download done.
File downloaded successfully
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08040000, size = 1473564
Download	[=========================] 100%      1473564 bytes
Download done.
File downloaded successfully
```

bryancostanich, 2019-11-20:
@clark-mcnally-pariveda can you jump onto [slack](http://slackinvite.wildernesslabs.co/) so we can diagnose this, please?

closed 2019-11-21 by clark-mcnally-pariveda