WildernessLabs/Meadow_Issues #41: Debug.Print() throws exception at runtime

LordBenjamin, 2019-11-20:
Calling `Debug.Print()` (as recommended in code samples in the developer documentation) causes an exception:

```c#
  public class MeadowApp : App<F7Micro, MeadowApp>
    {
        public MeadowApp()
        {
            Debug.Print("Hello Meadow!");
        }
    }
```

```
[0x4:] EXCEPTION handling: System.IO.FileNotFoundException: Could not load file or assembly 'System.Configuration, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies.
```
Calling `Console.WriteLine()` gives the correct output:

```c#
public class MeadowApp : App<F7Micro, MeadowApp>
    {
        public MeadowApp()
        {
            Console.WriteLine("Hello Meadow!");
        }
    }
```

```
Hello Meadow!
```

LordBenjamin, 2019-11-20:
I notice this could be related to #3.

bryancostanich, 2019-11-20:
Duplicate of #3

closed 2019-11-20 by bryancostanich