WildernessLabs/Meadow_Issues #46: Uploading CSharp.Scripting dlls causes Meadow to Lock up when dll is referenced

csuski, 2019-11-26:
Deployed the Microsoft.CodeAnalysis.CSharp.Scripting (3.3.1) libraries to the Meadow. Once this happens anything that references the libraries causes the Meadow to lock up.

Deploy the attached program to the Meadow. Verify all the Csharp.Scripting dlls get copied over.

The program blinks the onboard LED and will work fine.

However, if you try to redeploy the program, it should give an error:

"The semaphore timeout period has expired."

You should be able to reset the the device and it will blink the LED again (if you tried to redeploy you will need to reenable mono).

If you use the Meadow.CLI to try and list files it will lock up. The Meadow.CLI will eventually timeout without listing any files and if you watch the Meadow board it will have locked up also and stopped blinking the LED.

Another way to test is to uncomment the lines trying to use the scripting engine in the program. If you deploy with those lines uncommented it will lock up as soon as it tries to execute.

Once the Meadow is in this state the only way to fix it is to use the Meadow.CLI to reformat the file system.

[ScriptingBug.zip](https://github.com/WildernessLabs/Meadow_Issues/files/3893879/ScriptingBug.zip)

adrianstevens, 2020-01-04:
It's likely that the Microsoft.CodeAnalysis.CSharp.Scripting libraries have dependencies on core libraries that aren't currently being deployed to Meadow. I'd have to check the dependency tree but I suspect it's pretty deep. I also don't know if we'll be able to support loading code at runtime.

Meadow may also not have the hardware capabilities to run this package.

What are you trying to accomplish?

csuski, 2020-01-04:
I am not surprised it didn't work, but it was extremely difficult to recover from the error, or I guess it was hard to figure out how to recover from it, now that I know how it isn't much of an issue. I believe this is the same issue encountered here: https://github.com/WildernessLabs/Documentation/issues/138

The thing I found really odd, was that just loading the dll onto the board and then trying to list the files cause it to lock up even when that dll wasn't being used by the program, which made me think maybe there was more to the issue that it just not having the right references.

---

For the 2nd part of the question, I have been playing with a Ada Fruit Circuit Playground and using the Circuit Python stuff they designed for it and they have done some amazing things with it. One of them being the REPL interaction. 

I was trying to create a proof of concept of a some kind of REPL or runtime code interaction.

bryancostanich, 2020-12-05:
@adrianstevens is this fixed now with the new extension deps stuff?

adrianstevens, 2021-06-26:
This should be fixed with B5.1

closed 2021-07-07 by bryancostanich