WildernessLabs/Meadow_Issues #47: FC28.Read returns 0

tipa, 2019-12-03:
```
    public class MeadowApp : App<F7Micro, MeadowApp>
    {
        FC28 moistureSensor;

        public MeadowApp()
        {
            moistureSensor = new FC28(Device, Device.Pins.A00, Device.Pins.D00, 1.42f, 3.3f );
            moistureSensor.Updated += (s, e) => Console.WriteLine("Moisture update: " + e.New);
            moistureSensor.StartUpdating();

            Run();
        }

        async Task Run()
        {
            while (true)
            {
                float moisture = await moistureSensor.Read();
                Console.WriteLine($"Moisture while loop: {0}", moisture);
                Thread.Sleep(1000);
            }
        }
    }
```
The `Updated` event returns meaningful values while the manual `Read` calls return 0.
Experienced a similar behavior with the `Capacitive` moisture sensor.

```
Moisture while loop: 0
Moisture update: 0
Moisture while loop: 0
Moisture update: 0.003856928
Moisture while loop: 0
Moisture update: 0.003685469
Moisture while loop: 0
Moisture update: 0.004028386
Moisture while loop: 0
Moisture update: 0.1981162
Moisture while loop: 0
Moisture update: 1.060777
Moisture while loop: 0
Moisture update: 1.109845
Moisture while loop: 0
Moisture update: 1.12043
Moisture while loop: 0
Moisture update: 1.10556
Moisture while loop: 0
Moisture update: 1.095446
Moisture while loop: 0
Moisture update: 1.088118
Moisture while loop: 0
Moisture update: 1.083404
Moisture while loop: 0
Moisture update: 0.004671102
Moisture while loop: 0
Moisture update: 0.004585373
Moisture while loop: 0
Moisture update: 0.003985522
Moisture while loop: 0
Moisture update: 0.005099748
Moisture while loop: 0
Moisture update: 0.004628364
Moisture while loop: 0
Moisture update: 0.005271207
Moisture while loop: 0
```

adrianstevens, 2019-12-24:
@tipa this driver should be working correctly with the latest release of Meadow (b3.6)

adrianstevens, 2020-01-04:
The fix has been committed and the the nuget is published - closing issue

Thanks @tipa for reporting the bug!

closed 2020-01-04 by adrianstevens