WildernessLabs/Meadow_Issues #49: Unable to do a second deploy

csuski, 2019-12-23:
Updated to Meadow OS 0.3.5 With a December 22 build date and Visual Studio Extension 0.5.1 on Windows.

Deployed the sample projected from the VS template and everything when fine.  

Tried to deploy again and got the following Build output:

> 1>------ Deploy started: Project: MeadowApplication2, Configuration: Debug Any CPU ------
1>Deploying to Meadow on COM3...
1>Initializing Meadow
1>Checking files on device (may take several seconds)
1>Found App.exe
1>Found Meadow.dll
1>Found Meadow.Foundation.dll
1>Found GraphicsLibrary.dll
1>Found TftSpi.dll
1>Found mscorlib.dll
1>Found System.Core.dll
1>Found System.dll
1>Removing GraphicsLibrary.dll
1>Removing TftSpi.dll
1>Writing App.exe
1>The semaphore timeout period has expired.

If I try and deploy again I immediately get the semaphore error.  If I reset I can get it to do the same thing.

If I use the Meadow.CLI (Downloaded 12/23/19, 1.0 on build number) I can list files and get the following result:

> Opening port 'COM3'...ok.
Getting list of a files
Data: Mono is currently enabled to run applications
File List:
/meadow/App.exe
/meadow/Meadow.dll
/meadow/Meadow.Foundation.dll
/meadow/mscorlib.dll
/meadow/System.Core.dll
/meadow/System.dll

However, in Visual studio it still sees the original list.

If I try to delete a file with the CLI it reports:

> Opening port 'COM3'...ok.
Deleting

But after this no more commands work on the CLI everything command just returns with no info.  This seems to be the issue a lot of others are having.  After resetting the Meadow, I can list files again, and work with it, but the file is not deleted.

If I try and write a file with the CLI I get the following:

> Opening port 'COM3'...ok.
Writing .\Meadow2.dll
An exception was caught: System.IO.IOException: The semaphore timeout period has expired.
   at System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
   at System.IO.Ports.SerialStream.EndWrite(IAsyncResult asyncResult)
   at System.IO.Ports.SerialStream.Write(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
   at System.IO.Ports.SerialPort.Write(Byte[] buffer, Int32 offset, Int32 count)
   at MeadowCLI.Hcom.SendTargetData.EncodeAndSendPacket(Byte[] messageBytes, Int32 messageOffset, Int32 messageSize) in C:\Program Files (x86)\Jenkins\workspace\Meadow CLI\Meadow.CLI.Core\Internals\MeadowComms\SendTargetData.cs:line 261
   at MeadowCLI.Hcom.SendTargetData.BuildAndSendDataPacketRequest(Byte[] messageBytes, Int32 messageOffset, Int32 messageSize, UInt16 seqNumb) in C:\Program Files (x86)\Jenkins\workspace\Meadow CLI\Meadow.CLI.Core\Internals\MeadowComms\SendTargetData.cs:line 107
Timeout communicating with Meadow

csuski, 2019-12-23:
The issue seems to be that the MonoDisable command does not work once the App is running. This means you have to deploy or stop mono via the CLI immediately after restarting.

This was not the case in the previous release.

csuski, 2019-12-24:
More info - It is the new sample app that is the issue. For some reason you can't deploy while the sample app is running.  

The other apps I have made I can deploy changes while they are running, but with the sample app I have to reset and deploy before it starts the app or it will fail.

adrianstevens, 2019-12-24:
@csuski - does it work if you reset the board before deploying?

csuski, 2020-01-01:
It works if I reset the board and then deploy in the first 30 seconds or so after resetting.

adrianstevens, 2020-01-04:
Interesting, VS for Mac has no trouble deploying when the sample app is running. 

Unfortunately if the app is blocked or has crashed, Meadow won't respond to commands sent by Visual Studio to deploy. Resetting the board is good practice but we should make that clear in the docs.

Thanks @csuski

bryancostanich, 2020-02-01:
Fix in progress. Should be done for `b3.7` duplicate of #58

closed 2020-02-01 by bryancostanich