WildernessLabs/Meadow_Issues #51: EventHandler possibly returns at first Console.WriteLine

KiwiBryn, 2019-12-27:
While debugging an event handler I noticed that none of my code after the first Console.Writeline was executed. 

Normally I wouldn't have any Console.WriteLine statements in an event handler but without a debugger my options are limited.

This was code that had previously been used on Windows 10 IoT Core but with Debug.Writeline rather than Console.WriteLine.

[MeadowBinaryEvent.zip](https://github.com/WildernessLabs/Meadow_Issues/files/4004805/MeadowBinaryEvent.zip)

KiwiBryn, 2020-01-03:
I'm seeing the same issue with Event Handlers in Meadow.Foundation samples. I'm using the SHT31 Nuget from Meadow.Foundation.Sensors.Atomespheric

'App.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_64\mscorlib\v4.0_4.0.0.0__b77a5c561934e089\mscorlib.dll'. 
'App.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'C:\Users\BrynLewis\source\repos\LoRaFieldGatewayMeadowClient\SHT31TestHarness\bin\Debug\net472\App.exe'. Symbols loaded.
'App.exe' (CLR v4.0.30319: App.exe): Loaded 'C:\Users\BrynLewis\source\repos\LoRaFieldGatewayMeadowClient\SHT31TestHarness\bin\Debug\net472\Meadow.dll'. 
The program '[19984] App.exe: Program Trace' has exited with code 0 (0x0).
The program '[19984] App.exe' has exited with code 0 (0x0).
 .
 .
 .
 .
 DirectRegisterAccess = True
 Initialising sensor...
 .
 Temp: 0
 Humidity: 0
 Temp: 22.97322
 Temp: 22.97322
 Temp: 22.95987
 Temp: 22.97322
 Temp: 22.97322

When I download the binary in the output window 
1>------ Build started: Project: SHT31TestHarness, Configuration: Debug Any CPU ------
1>SHT31TestHarness -> C:\Users\BrynLewis\source\repos\LoRaFieldGatewayMeadowClient\SHT31TestHarness\bin\Debug\net472\App.exe
2>------ Deploy started: Project: SHT31TestHarness, Configuration: Debug Any CPU ------
2>Deploying to Meadow on COM3...
2>Initializing Meadow
2>Device  **MeadowOS Version: 0.3.5** (Dec 22 2019 05:20:04)
2>Checking files on device (may take several seconds)

This is with the latest MeadowOS download files dated 24 Dec 2019 11:34:56 am which seems a bit odd

[SHT31TestHarness.zip](https://github.com/WildernessLabs/Meadow_Issues/files/4018575/SHT31TestHarness.zip)

adrianstevens, 2020-01-04:
Thx @KiwiBryn - I'll test over the weekend 

And glad to see you hacking on Meadow!

KiwiBryn, 2020-01-04:
I have code in my [LoRa field gateway client] (https://github.com/KiwiBryn/FieldGateway.LoRa.MeadowClient) where tried  updating an LED's state after a call to Console.WriteLine in an event handler. 

private void Rfm9XDevice_OnTransmit(object sender, Rfm9XDevice.OnDataTransmitedEventArgs e)
{
   Console.WriteLine("{0:HH:mm:ss}-TX Done 1", DateTime.UtcNow);
   Led.IsOn = false;
   Console.WriteLine("{0:HH:mm:ss}-TX Done 2", DateTime.UtcNow);
}

 Register 0x41 - Value 0X00 - Bits 00000000
 Register 0x42 - Value 0X12 - Bits 00010010
 12:00:35-TX T:23.5C H:47%
 00:00:46-TX Done 1
 12:01:36-TX T:23.5C H:46%
 00:01:36-TX Done 1
 12:02:36-TX T:23.5C H:46%
 00:02:37-TX Done 1

The LED flashes as expected so I think it be the first Console.WriteLine works then the following one(s) fail silently.

Thanks
@KiwiBryn

ctacke, 2020-02-01:
Probably [related to this bug](https://github.com/WildernessLabs/Meadow_Issues/issues/57)

closed 2020-02-09 by bryancostanich