WildernessLabs/Meadow_Issues #53: Calling ToString on Stopwatch.Elapsed appears to halt application

thegreatco, 2020-01-06:
I was trying to print the uptime to the console as a check the app was still running and it just seems to hard lock (sometimes requires hitting the reset button to deploy a new app) whenever I try to format the stopwatch output. I haven't tried this with any other `TimeSpan` or `DateTime`, so I don't know if it is unique to `Stopwatch`.

Simple repro:

```
public class Test : App<F7Micro, Test>
{
    private readonly Stopwatch _stopwatch;
    public Test()
    {
        _stopwatch = Stopwatch.StartNew();
        PrintUptime();
    }
    private void PrintUptime()
    {
        while(true)
        {
            Console.WriteLine("About to print uptime...");
            Console.WriteLine(_stopwatch.Elapsed.ToString("hh:mm:ss"));
            Thread.Sleep(1000);
        }
    }
}
```

bryancostanich, 2020-01-06:
Can you post your `static void main()` loop, as well, please?

thegreatco, 2020-01-06:
@bryancostanich updated.

bryancostanich, 2020-01-06:
Add `Thread.Sleep(Timeout.Infinite)` to the end of your `main()`; it’s returning and your app is exiting.

thegreatco, 2020-01-06:
The `Thread.Sleep(Timeout.Infinite)` got the exception to bubble up. I think that `Main()` came from the template when I created my first app, just copied it around. I guess the other issue was on me, `TimeSpan` format strings are not the same as `DateTime` format strings.

```
Console.WriteLine(_stopwatch.Elapsed.ToString(@"hh\:mm\:ss"));
Console.WriteLine(DateTime.UtcNow.ToString("hh:mm:ss"));
```

closed 2020-01-06 by thegreatco