WildernessLabs/Meadow_Issues #54: Configurable Meadow device names

KiwiBryn, 2020-01-12:
Using Environment.MachineName I can get the name of a device. (?It's always "Mono"). 

With .NetMF used to be able to set the device name with MFDeploy as part of a provisioning process. 

It would be really useful to be able to configure the Meadow device name for provisioning/inventory management etc. from either the CLI or VisualStudio tooling.

If this could also be the Network name of the device so we can see it in DHCP console etc. 

@KiwiBryn

adrianstevens, 2020-01-16:
@KiwiBryn great idea imo - we're still working through a few "big rocks" on the OS side but I suspect this would make the list as we get closer to v1.0

bryancostanich, 2020-01-31:
This is a great feature request. I'm going to set it to b4.0 timeframe and see if we can get it in.

bryancostanich, 2020-06-05:
@Peter-Moody let's talk about this the next engineering meeting.

NevynUK, 2021-08-09:
This will be configurable through a config file on the device.

closed 2021-11-08 by NevynUK