WildernessLabs/Meadow_Issues #55: Feature request : SPI frequency change

MikroBusNet, 2020-01-20:
Hello,

Some devices need to initialized at low frequency and then be set to higher one for exploitation. This is not possible with Meadow.

Could it be possible to have either a setter or a method for the SPI frequency setting ?

Best regards,

ctacke, 2020-02-07:
Beta 3.7 allows you to change configuration after construction, so changing frequency or polarity is allowed.

MikroBusNet, 2020-02-19:
This one should not be closed as it is not fixed in Beta 3.7.

Members of the SpiConfiguration class are still read-only.

ctacke, 2020-02-22:
Tested this code against upcoming beta 3.8, and the scope is definitely reflecting clock frequency changes.

`while (true)
{
    // this is purely a test for supporting changing bus speed in SPI - remove when the radio is implemented
    foreach (var spd in bus.SupportedSpeeds)
    {
        Console.WriteLine($"{spd}");

        try
        {
            bus.Configuration.SpeedKHz = spd;

            Console.WriteLine($" @{bus.Configuration.SpeedKHz} Silicon Revision: 0x{radio.GetVersion():x2}");
        }
        catch(Exception ex)
        {
            Console.WriteLine($" {ex.Message}");
        }

        Thread.Sleep(2000);
    }
}
`

bryancostanich, 2020-03-09:
b3.8

closed 2020-03-09 by bryancostanich