WildernessLabs/Meadow_Issues #62: Constant 10ms delay when calling Thread.Sleep()

MikroBusNet, 2020-02-18:
Hello,

Running the following code with bet 3.7, we can see a constant extra delay of 10ms added to the sleep duration (see screenshot).
```
private void TestSleep()
        {
            var _pin = Device.CreateDigitalOutputPort(Device.Pins.D04);

            while (true)
            {
                Thread.Sleep(50);
                _pin.State = true;
                Thread.Sleep(50);
                _pin.State = false;
            }
        }
```
Result is a pulse width of 60ms and, of course, a period of 120ms.
The exact same code with NETMF on a F427 board (Quail) gives 50.10 and 100.20 respectively. (see screenshot again).

I say constant 10ms delay because no matter which value you put for the sleep delay, there will always be 10ms added to that value.

Such constant is surprising. It was already present in the beta 3.6 which was "slower" than beta 3.7 on many aspects. So this is not tied to overall performance, to me.

Meadow timings :

[image]

NETMF timings :

[image]

bryancostanich, 2020-04-02:
FYI: This is because the time resolution in our NuttX config is low. We're going to increase the resolution, but there some issues with that we need to work through. Not sure which release it'll make it into. Just wanted to let you know we validated and understand the issue.

bryancostanich, 2020-05-26:
Can you run this again and tell us what the delay is now? We changed our tick resolution in NuttX from 10ms to 1ms. Might improve this.

ctacke, 2020-06-19:
Beta-4 has a pulse width of 51ms.  We've changed the OS scheduler tick to 1ms, and unless that changes it's unlikely to get any closer to 50.  We wait 50, then on the next tick the output is set, giving the resultant 51.

MikroBusNet, 2020-06-19:
Then why not wait 49 ?

closed 2020-07-15 by bryancostanich