WildernessLabs/Meadow_Issues #63: Beta 3.7 - I2C frequencies not correct

MikroBusNet, 2020-02-23:
When compared to NETMF 4.4, Meadow I2C frequencies are far from the expected values.

Here are some measures :

- NETMF 100KHz : 100 KHz measured
- Meadow 100KHz : 95.7 to 96.4 KHz measured (**4% difference** from expected value)
- NETMF 400KHz : 393.7 to 397.3 KHZ measured (**1.2% difference** from expected value)
- Meadow 400KHz : 362.3 to 367.6 KHz measured **(8.8% difference** from expected value)

Difference are quite  significant, to me.

ctacke, 2020-02-24:
The I2C Spec, if I recall correctly, is a "not to exceed" type, so if you set it at 400KHz, the speed should be <= the requested speed.  While I agree that these are off (and by a fair bit on the 400K) are they causing an actual peripheral problem for you?  This will help us prioritize this.

I just looked at the underlying Nuttx driver that gets called, and it's looking specifically for values of 100000, 400000 and 1000000.  Any other value falls into an "else" condition  that is probably 10k.  The value selected sets up a prescaler then some low and high values.  These are all likely dependent on the system clock, and we may be able to tune them to get closer to the exact values, but again, knowig how to prioritize this will help.

MikroBusNet, 2020-02-24:
I recall the same about the spec.
Then, regardless of the frequency accuracy, there are problems with the implementation in the Nuttx driver :
- the fallback is not the one it should be : if I try to set 500KHz, it should fall back to 400 KHz, not 10KHz
- there is still an issue with the exact value of 1000000 that gives an "_ioctl I2CData failed TryAgain_"

Now, to be fair, it did not cause any issues so far with the peripherals I've tested. Perhaps there is some kind of tolerance ? But with such disparity, we are not immune to some issues at some time.

Anyway, this is clearly not a high priority.

bryancostanich, 2020-03-01:
> the fallback is not the one it should be : if I try to set 500KHz, it should fall back to 400 KHz, not 10KHz

@MikroBusNet is that what's happening today?

> there is still an issue with the exact value of 1000000 that gives an "ioctl I2CData failed TryAgain"

@ctacke can you look into that, please?

MikroBusNet, 2020-03-01:
> @MikroBusNet is that what's happening today?
Yes. Every value not being exactly 100K or 400K falls back to a resulting 10KHz frequency.

MikroBusNet, 2020-03-04:
I still think that the easiest and safest way would be to use an enum, like this one :
```
public enum ClockFrequency
    {
        /// <summary>
        /// Standard clock speed with a clock rate of 100 KHz.
        /// </summary>
        Standard = 100000,

        /// <summary>
        /// Fast clock speed with a clock rate of 400 KHz.
        /// </summary>
        Fast = 400000,

        /// <summary>
        /// Hig Speed with a clock rate of 1 MHz.
        /// </summary>
        FastPlus = 1000000,

        /// <summary>
        /// Hig Speed with a clock rate of 3.4 MHz.
        /// </summary>
        HighSpeed = 3400000,

        /// <summary>
        /// Hig Speed with a clock rate of 5 MHz (unidirectionnal mode).
        /// </summary>
        UltraFast = 5000000
    }
```
In the underlying code, a simple cast (to an Int32) would enough to set the right frequency. No tests, no roundings, nothing prone to issues.
But you decide ;)

bryancostanich, 2020-03-09:
b3.8

closed 2020-03-09 by bryancostanich