WildernessLabs/Meadow_Issues #68: Static class in its own namespace hangs in constructor

MikroBusNet, 2020-03-06:
Take the following static class, in the MeadowApplication2 namespace :
```
using Meadow;
using Meadow.Devices;
using Meadow.Hardware;
namespace MeadowApplication2
{
    public partial class MeadowApp : App<F7Micro, MeadowApp>
    {
        public static class StaticClass
        {
            public static readonly IDigitalOutputPort Led1;
            static StaticClass()
            {
                Led1 = Device.CreateDigitalOutputPort(Device.Pins.OnboardLedGreen);
            }
        }
    }
}

```
Using it in the following program does work and the green led is on :
```
using System;
using System.Threading;
using Meadow;
using Meadow.Devices;
using static System.Console;
namespace MeadowApplication2
{
    public partial class MeadowApp : App<F7Micro, MeadowApp>
    {
        public MeadowApp()
        {
            WriteLine($"{DateTime.Now} : Starting program");
            WriteLine($"{DateTime.Now} : Led1 ON");
            StaticClass.Led1.State = true;
            WriteLine($"{DateTime.Now} : Going to sleep...");
            Thread.Sleep(Timeout.Infinite);
        }
    }
}

```

Now, if I put the static class in its own namespace, like this :
```
using Meadow;
using Meadow.Devices;
using Meadow.Hardware;
namespace MyNamespace
{
    public partial class MeadowApp : App<F7Micro, MeadowApp>
    {
        public static class StaticClass
        {
            public static readonly IDigitalOutputPort Led1;
            static StaticClass()
            {
                Led1 = Device.CreateDigitalOutputPort(Device.Pins.OnboardLedGreen);
            }
        }
    }
}
```
Then the following code hangs in the static constructor :
```
using System;
using System.Threading;
using Meadow;
using Meadow.Devices;
using static System.Console;
namespace MeadowApplication2
{
    public partial class MeadowApp : App<F7Micro, MeadowApp>
    {
        public MeadowApp()
        {
            WriteLine($"{DateTime.Now} : Starting program");
            WriteLine($"{DateTime.Now} : Led1 ON");
            MyNamespace.MeadowApp.StaticClass.Led1.State = true;
            WriteLine($"{DateTime.Now} : Going to sleep...");
            Thread.Sleep(Timeout.Infinite);
        }
    }
}
```

ctacke, 2020-06-19:
Your code contains two `App<D,A>` classes.  We'd not considered that ever happening and there was a race condition in the constructor accessing a static property.  This is fixed in Beta 4, but I'd still encourage you to look at an using an architecture that contains one and only one App class.

MikroBusNet, 2020-06-19:
If you look carefuly, you will see that there is only one **partial** `App<D, A>` class.

closed 2020-07-15 by bryancostanich