WildernessLabs/Meadow_Issues #69: Timer creep

agillesp, 2020-03-07:
`System.Threading.Timer` and `System.Timers.Timer` both experience creep. Configure an event to fire every second and it'll creep by about 1/100th a second each tick.

[image]

agillesp, 2020-03-07:
Maybe related to #62

MikroBusNet, 2020-03-07:
I confirm this behaviour.

And System.timer has that constant 10ms  added as well. Setting a period of 50ms is measured as 60ms on the oscilloscope.

bryancostanich, 2020-04-02:
FYI: This is because the time resolution in our NuttX config is low. We're going to increase the resolution, but there some issues with that we need to work through. Not sure which release it'll make it into. Just wanted to let you know we validated and understand the issue.

ctacke, 2020-06-19:
Pre-beta4 has this down to approximately 3ms latency.  If your application has tight timing requirements, you can account for this in code, however bear in mind this is all managed code, so a GC or whatever is likely to make this not 100% reliable.

I did this:
```
private void TestTimer()
{
    new Timer(myTimerCallBack).Change(0, 997);
    Thread.Sleep(Timeout.Infinite);
}

private void myTimerCallBack(object state)
{
    Console.WriteLine($"{DateTime.Now:HH:mm:ss.fff}");
}
```
And see this output:
```
App: 01/01/1970 00:00:14 : Application starting
App: 00:00:18.444
App: 00:00:19.108
App: 00:00:20.106
App: 00:00:21.106
App: 00:00:22.106
App: 00:00:23.106
App: 00:00:24.106
App: 00:00:25.106
App: 00:00:26.106
App: 00:00:27.106
App: 00:00:28.106
App: 00:00:29.106
App: 00:00:30.106
App: 00:00:31.106
App: 00:00:32.106
App: 00:00:33.106
App: 00:00:34.106
App: 00:00:35.106
App: 00:00:36.106
App: 00:00:37.106
App: 00:00:38.106
App: 00:00:39.106
```

closed 2020-07-15 by bryancostanich