WildernessLabs/Meadow_Issues #70: Assertion at sgen-stw.c:69

MikroBusNet, 2020-03-08:
After some time (not really measured, but roughly between 10 and 15 min), the code below is displaying this error message in the VS console :

>  * Assertion at sgen-stw.c:69, condition `info->client_info.stack_start >= info->client_info.info.stack_start_limit && info->client_info.stack_start < info->client_info.info.stack_end' not met
> mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error code '142'" (142)
>  mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error code '142'" (142)
>  mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error code '142'" (142)

Code :

```
using Meadow;
using Meadow.Devices;
using Meadow.Foundation;
using Meadow.Foundation.Leds;
using Meadow.Hardware;
using System;
using System.Diagnostics;
using System.Threading;
namespace TestResources
{
    public class MeadowApp : App<F7Micro, MeadowApp>
    {
        IDigitalOutputPort dio;
        Boolean D13State;
        public MeadowApp()
        {
            Console.WriteLine($"{DateTime.Now} : Application starting");
            TestTimer();
        }
        private void TestTimer()
        {
            dio = Device.CreateDigitalOutputPort(Device.Pins.D13, false);
            new Timer(myTimerCallBack).Change(0, 50);
        }
        private void myTimerCallBack(object state)
        {
            D13State = !D13State;
            dio.State = D13State;
        }
    }
}
```

ctacke, 2020-06-19:
I do not see the assertion output, but I can confirm that this crash still occurs in the current (June 19, 2020) pre-b4 build.

bryancostanich, 2020-07-06:
Fixed with the new GC work from Kangaroo!

bryancostanich, 2020-07-06:
will be released in `b4.0`

closed 2020-07-15 by bryancostanich