WildernessLabs/Meadow_Issues #72: Beta 3.8 - Interrupt not firing

MikroBusNet, 2020-03-12:
The following code does not output anything when 3.3V is applied to the pin D05. It was working with 3.7.

```
IDigitalInterruptPort Button;
        public MeadowApp()
        {
            Button = Device.CreateDigitalInputPort(Device.Pins.D05, InterruptMode.EdgeBoth, ResistorMode.Disabled);
            Button.Changed += Button_Changed;
        }

        private void Button_Changed(Object sender, DigitalInputPortEventArgs e)
        {
                Console.WriteLine($"Button pressed : {e.Value}, {e.New}");
        }
```

MikroBusNet, 2020-03-15:
For reference, please see : https://app.slack.com/client/TB8K9QWNR/CQWD660E8/thread/CQWD660E8-1584048792.076800

adrianstevens, 2020-03-28:
Fixed in beta 3.9 - closing

Thanks @MikroBusNet for reporting!

closed 2020-03-28 by adrianstevens