WildernessLabs/Meadow_Issues #73: Beta 3.7 - Interrupt event freezes

MikroBusNet, 2020-03-12:
The following code does not display anything when 3.3V is applied to the pin D05.
But if you remove `{e.Old}` and `{e.Delta}` from the WriteLine() then you can see interrupt firing.

```
IDigitalInterruptPort Button;
        public MeadowApp()
        {
            Button = Device.CreateDigitalInputPort(Device.Pins.D05, InterruptMode.EdgeBoth, ResistorMode.Disabled);
            Button.Changed += Button_Changed;
        }

        private void Button_Changed(Object sender, DigitalInputPortEventArgs e)
        {
                Console.WriteLine($"Button pressed : {e.Value}, {e.Old}, {e.New}, {e.Delta}");
        }
```
No exception is generated. At least, a try/catch block does not show anything.

adrianstevens, 2020-03-28:
@MikroBusNet this should be working correctly now in the latest release (beta 3.9)

bryancostanich, 2020-04-04:
@adrianstevens is this fixed? You added a `b3.9` label to it.

Biswis, 2020-05-21:
I'm running on the  b3.10 OS and I'm not sure it's been fixed.  I have a button (with pull-down) attached to D15 (also tried D05) and running this code:

```
    public class ButtonEventsApp : App<F7Micro, ButtonEventsApp>
    {
        IDigitalInputPort _input;

        public ButtonEventsApp()
        {
            _input = Device.CreateDigitalInputPort(Device.Pins.D15);
            _input.Changed += Input_Changed;

            while( true)
            {
                Console.WriteLine(_input.State.ToString());
                Thread.Sleep(1000);
            }
        }

        private void Input_Changed(object sender, DigitalInputPortEventArgs e)
        {
            Console.WriteLine("Changed: " + e.Value.ToString() + ", Time: " + e.New.ToString());
        }
    }
```

I see the state of the button press change in the output but the change event doesn't appear to fire:

>  False
>  False
>  True
>  True
>  True
>  True
>  False

Biswis, 2020-05-22:
After the demos today I figured out that I hadn't initialised the port correctly.  Needs to be:

`            _input = Device.CreateDigitalInputPort(
                                                  Device.Pins.D15, 
                                                  InterruptMode.EdgeRising, 
                                                  ResistorMode.Disabled);`

And now it is working as expected.

The InterruptMode is "None" by default so the interrupts and therefore the change events don't fire presumably?

bryancostanich, 2020-06-05:
Should be all fixed up now.

closed 2020-06-05 by bryancostanich