WildernessLabs/Meadow_Issues #74: Interrupts : long delay before first signal

MikroBusNet, 2020-03-13:
When using interrupts, the first interrupt takes roughly 3 seconds (!) before being fired. After that, every interrupt is fired almost instantly.

I am using the following code, tying D04 to D05, D05 being the interrupt port. Doing this way, there is no bouncing to handle.

```
public class MeadowApp : App<F7Micro, MeadowApp>
    {
        IDigitalInterruptPort _receiver;
        IDigitalOutputPort _sender;

        public MeadowApp()
        {
            _sender = Device.CreateDigitalOutputPort(Device.Pins.D04, false);
// Triggering the output pin before enabling interrupt port
// This ensures that the delay is not caused by triggering the output pin for the first time
            _sender.State = true;
            _sender.State = false;

            _receiver = Device.CreateDigitalInputPort(Device.Pins.D05, InterruptMode.EdgeBoth, ResistorMode.Disabled);
            _receiver.Changed += _receiver_Changed;

            Console.WriteLine($"{DateTime.Now} : pressing button and waiting 1s");
            _sender.State = true;
            Thread.Sleep(1000);

            Console.WriteLine($"{DateTime.Now} : releasing button and waiting 1s");
            _sender.State = false;
            Thread.Sleep(1000);

            Console.WriteLine($"{DateTime.Now} : pressing button and waiting 1s");
            _sender.State = true;
            Thread.Sleep(1000);

            Console.WriteLine($"{DateTime.Now} : releasing button");
            _sender.State = false;
        }

        private void _receiver_Changed(Object sender, DigitalInputPortEventArgs e)
        {
            Console.WriteLine($"{DateTime.Now} : button event, state = {e.Value}");
        }
    }
```

Console output :

>  01/01/1970 00:00:13 : pressing button and waiting 1s
>  01/01/1970 00:00:16 : button event, state = True
>  01/01/1970 00:00:17 : releasing button and waiting 1s
>  01/01/1970 00:00:17 : button event, state = False
>  01/01/1970 00:00:18 : pressing button and waiting 1s
>  01/01/1970 00:00:18 : button event, state = True
>  01/01/1970 00:00:19 : releasing button
>  01/01/1970 00:00:19 : button event, state = False

I am using 3.7 because of issues with 3.8 and interrupts but I think that 3.8 could exhibit the same behaviour.

ctacke, 2020-06-19:
Tested and confirmed. See issue 75 for more details on solution and timeline.
https://github.com/WildernessLabs/Meadow_Issues/issues/75

bryancostanich, 2020-07-11:
Just to clarify, this is because the interpreter "hot-paths" the code and caches it. It's very similar to JIT; you can think of it as like a per-method JIT. And Chris is correct, AOT will get rid of the initial hit. Added the work item label.

ctacke, 2022-08-20:
There's still a delay on first interrupt, but it's much less noticeable now.  With JIT enabled, it's between 10 and 20ms.

patridge, 2022-10-27:
Closing as resolved now that [RC-1 is released](http://developer.wildernesslabs.co/Meadow/Release_Notes/Release-Candidates/).

closed 2022-10-27 by patridge