WildernessLabs/Meadow_Issues #75: Long delay when creating IO port for the first time

MikroBusNet, 2020-03-13:
There is always a long delay when creating an IO port for the first time : 

```
public MeadowApp()
        {
            AppDomain currentDomain = AppDomain.CurrentDomain;
            currentDomain.UnhandledException += CurrentDomain_UnhandledException;

            Console.WriteLine($"Application starting");
            
            Console.WriteLine($"{DateTime.Now} : Creating Output port d1");
            var d1 = Device.CreateDigitalOutputPort(Device.Pins.D01);
            Console.WriteLine($"{DateTime.Now} : d1 Output port created");

            Console.WriteLine($"{DateTime.Now} : Creating Output port d2");
            var d2 = Device.CreateDigitalOutputPort(Device.Pins.D02);
            Console.WriteLine($"{DateTime.Now} : d2 Output port created");

            Console.WriteLine($"{DateTime.Now} : Creating input port d3");
            var d3 = Device.CreateDigitalInputPort(Device.Pins.D03);
            Console.WriteLine($"{DateTime.Now} : d3 input port created");

            Console.WriteLine($"{DateTime.Now} : Creating input port d4");
            var d4 = Device.CreateDigitalInputPort(Device.Pins.D04);
            Console.WriteLine($"{DateTime.Now} : d4 input port created");
        }
```

Console output :

>  Application starting
>  01/01/1970 00:00:11 : Creating Output port d1
>  01/01/1970 00:00:15 : d1 Output port created
>  01/01/1970 00:00:15 : Creating Output port d2
>  01/01/1970 00:00:15 : d2 Output port created
>  01/01/1970 00:00:15 : Creating input port d3
>  01/01/1970 00:00:15 : d3 input port created
>  01/01/1970 00:00:15 : Creating input port d4
>  01/01/1970 00:00:15 : d4 input port created

This is probably related to issue #74

ctacke, 2020-06-19:
Tested and confirmed in pre-release b4.0 code.

This behavior is not a surprise since AoT is not yet supported - the code path in Meadow.Core is being JITted on first object creation and simply takes time.  For now the recommendation is that this is known behavior and applications should always take first-run JIT time into account and plan accordingly.
My test code:

```
public void PortCreation()
{
    var pins = new IPin[]
    {
        Device.Pins.D01,
        Device.Pins.D02,
        Device.Pins.D03,
        Device.Pins.D04,
        Device.Pins.D05,
        Device.Pins.D06,
        Device.Pins.D07,
        Device.Pins.D08,
        Device.Pins.D09,
    };

    var results = new long[pins.Length];
    long start;

    for(int i = 0; i < pins.Length; i++)
    {
        start = Environment.TickCount;
        var p = Device.CreateDigitalOutputPort(pins[i]);
        results[i] = Environment.TickCount - start;
    }

    long sum = 0;
    for (int i = 0; i < results.Length; i++)
    {
        Console.WriteLine($"Port {i} took {results[i]}ms");
        sum += results[i];
    }
    Console.WriteLine($"mean: {(sum/(float)results.Length):0}ms");
}
```

bryancostanich, 2020-07-11:
Just to clarify, this is because the interpreter "hot-paths" the code and caches it. It's very similar to JIT; you can think of it as like a per-method JIT. And Chris is correct, AOT will get rid of the initial hit. Added the work item label.

bryancostanich, 2021-05-10:
Dupe of #74

closed 2021-05-10 by bryancostanich