WildernessLabs/Meadow_Issues #77: PWM duration with TimeScale.MicroSecond off by 10µs

MikroBusNet, 2020-03-15:
With a frequency of 1 KHz, and starting with duration at 30 µs, the pulse width is 10µs less than expected. Below 30µs, I don't measure any significant difference.

The following code :
```
var pwm = Device.CreatePwmPort(Device.Pins.D04, 1000, 0.030f, false);
pwm.Start();
Thread.Sleep(5000);

pwm.TimeScale = TimeScale.Microseconds;
pwm.Duration = 30.0f;
Thread.Sleep(5000);

pwm.TimeScale = TimeScale.Milliseconds;
pwm.Duration = 0.030f;
```
gives a pulse width measured at :

- TimeScale.Seconds : 30 µs duration, Duty cycle 3%/97%
- TimeScale.MicroSeconds : 20 µs duration, Duty cycle 2%/98%
- TimeScale.MilliSeconds : 30 µs duration, Duty cycle 3%/97%

Same for this one, at 100µs :
```
var pwm = Device.CreatePwmPort(Device.Pins.D04, 1000, 0.1f, false);
pwm.Start();
Thread.Sleep(5000);

pwm.TimeScale = TimeScale.Microseconds;
pwm.Duration = 100.0f;
Thread.Sleep(5000);

pwm.TimeScale = TimeScale.Milliseconds;
pwm.Duration = 0.1f;
```
Measured :

- TimeScale.Seconds : 100 µs duration, Duty cycle 10%/90%
- TimeScale.MicroSeconds : 90 µs duration, Duty cycle 9%/91%
- TimeScale.MilliSeconds : 100 µs duration, Duty cycle 10%/90%

We can see the same behaviour with other frequencies, the offset being of course different according to the frequency.
For example, at 2 KHz and a 10% duty cycle, we can measure : 

- TimeScale.Seconds : 50 µs duration, Duty cycle 10%/90%
- TimeScale.MicroSeconds : 45 µs duration, Duty cycle 9%/91%
- TimeScale.MilliSeconds : 50 µs duration, Duty cycle 10%/90%

bryancostanich, 2020-04-02:
FYI: This is because the time resolution in our NuttX config is low. We're going to increase the resolution, but there some issues with that we need to work through. Not sure which release it'll make it into. Just wanted to let you know we validated and understand the issue.

ctacke, 2020-06-19:
We had a math error in the code that was essentially locking the duty cycle to multiples of 10us. Fixed in the next release.

closed 2020-07-15 by bryancostanich