WildernessLabs/Meadow_Issues #86: b3.9 UART issues

StephenCardinale, 2020-03-30:
The following code throws an exception when you attempt to open a serial port:
`public MeadowApp()
    {
      ISerialPort _serial = Device.CreateSerialPort(Device.SerialPortNames.Com1, 9600);
      _serial.Open();
    }`

The exception is as follows:
`'App.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_64\mscorlib\v4.0_4.0.0.0__b77a5c561934e089\mscorlib.dll'.
'App.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'C:\Users\...\source\repos\TestSerial\TestSerial\bin\Debug\net472\App.exe'. Symbols loaded.
'App.exe' (CLR v4.0.30319: App.exe): Loaded 'C:\Users\...\source\repos\TestSerial\TestSerial\bin\Debug\net472\Meadow.dll'.
The program '[20844] App.exe' has exited with code 0 (0x0).
 Unhandled Exception: System.EntryPointNotFoundException: tcgetattr assembly:<unknown assembly> type:<unknown type> member:(null)
 at (wrapper managed-to-native) Meadow.Core.Interop+Nuttx.tcgetattr(intptr,Meadow.Core.Interop/Nuttx/Termios&)
 at Meadow.Hardware.SerialPort.SetPortSettings () <0xc02c0b20 + 0x00016> in <49c0748cb5f241238302eacf325ae8b9>:0
 at Meadow.Hardware.SerialP ort.Open () <0xc02bfd48 + 0x00066> in <49c0748cb5f241238302eacf325ae8b9>:0
 at TestSerial.MeadowApp..ctor () <0xc0105558 + 0x00048> in <00b629e6468c47cda1896774f1c14ff1>:0
 at TestSerial.Program.Main (System.String[] args) <0xc01051b8 + 0x00034> in <00b629e6468c47cda1896774f1c14ff1>:0
`

bryancostanich, 2020-04-02:
We haven't been able to reproduce this, but we've also made some fixes around this, so when we get `b3.10` out, can you test it on that, and see if it still happens, please?

vimmers2020, 2020-04-03:
I could reproduce this issue, unfortunately...

Code:
`            Console.WriteLine("Creating serial port COM4...");
            this.port = Device.CreateSerialPort(Device.SerialPortNames.Com4, 9600, 8, Parity.None, StopBits.One, 4096);
            Console.WriteLine("Opening serial port...");
            this.port.Open();
`
Messages in the console:
`Launching application...
 Creating serial port COM4...
 Opening serial port...
 Unhandled Exception: System.EntryPointNotFoundException: tcgetattr assembly:<unknown assembly> type:<unknown type> member:(null)
  at (wrapper managed-to-native) Meadow.Core.Interop+Nuttx.tcgetattr(intptr,Meadow.Core.Interop/Nuttx/Termios&)
  at Meadow.Hardware.SerialPort.SetPortSettings () <0xc0386428 + 0x00016> in <49c0748cb5f241238302eacf325ae8b9>:0 
  at Meadow.Hardware.SerialP ort.Open () <0xc037f1f8 + 0x00066> in <49c0748cb5f241238302eacf325ae8b9>:0 
  at GPSTestApp.GPSMeadowApp..ctor () <0xc0105ee0 + 0x000c8> in <ef8757922e79412294395404be19edd3>:0 
  at GPSTestApp.Program.Main (System.String[] args) <0xc01057d8 + 0x00034> in <ef8757922e79412294395404be19edd3>:0 `

Note: My code has been running without problem tens of times a week ago, but doesn't anymore now. I meanwhile did a lot of other things with the Meadow, so the issue might depend on the history of the board. Maybe a reason why it is hard to reproduce.

bryancostanich, 2020-04-03:
`b3.10` should be out soon which fixes this.

bryancostanich, 2020-04-04:
`b3.10` is out.

closed 2020-04-04 by bryancostanich