WildernessLabs/Meadow_Issues #87: SerialPort.GetPortNames(); Hangs Meadow

StephenCardinale, 2020-03-30:
The following code will hang the Meadow and it will become unresponsive to subsequent deployments without power cycling the Meadow.
`String[] names = SerialPort.GetPortNames();
 Console.WriteLine($"Available Serial Ports is {names.Length}");
 
foreach (String t in names)
{
       Console.WriteLine($"{t} is available as a COMM Port on the Meadow.");
}`

tritao, 2020-04-02:
Tried to reproduce this and get this output with the sample code.

```
App: ttyACM0 is available as a COMM Port on the Meadow.
App: ttyS0 is available as a COMM Port on the Meadow.
App: ttyS1 is available as a COMM Port on the Meadow.
App: ttyS2 is available as a COMM Port on the Meadow.
```

rudizz, 2020-04-04:
I have the same problem.
I also get an exception after I call ISerialPort.Open. Here's the code:
```
var sp4 = Device.CreateSerialPort(Device.SerialPortNames.Com4, 115200);
Console.WriteLine("Create sp4...");
sp4.Open();
Console.WriteLine("Open sp4...");
```
After "Create sp4..." I receive:
```
Unhandled Exception: System.EntryPointNotFoundException: tcgetattr assembly:<unknown assembly> type:<unknown type> member:(null)
at (wrapper managed-to-native) Meadow.Core.Interop+Nuttx.tcgetattr(intptr,Meadow.Core.Interop/Nuttx/Termios&)
  at Meadow.Hardware.SerialPort.SetPortSettings () <0xc03579f8 + 0x00016> in <49c0748cb5f241238302eacf325ae8b9>:0 
  at Meadow.Hardware.SerialP ort.Open () <0xc034de18 + 0x00066> in <49c0748cb5f241238302eacf325ae8b9>:0 
  at MeadowApp.MeadowApp.Initialize () <0xc02dba00 + 0x0005a> in <44203868a894412789252287fbdc9988>:0 
  at MeadowApp.MeadowApp..ctor () <0xc0105900 + 0x0000c> in <44203868a894412789252287fbdc9988>:0 
  at MeadowApp.Program.Main (System.String[] args) <0xc0105760 + 0x00000> in <44203868a894412789252287fbdc9 988>:0 

```
I'm facing this problem in Windows as well as in MacOS.

tritao, 2020-04-04:
@rudizz Can you try the latest Beta 3.10 release?

I think it fixes that particular error.

rudizz, 2020-04-04:
@tritao with Beta 3.10 it works.
Thanks

bryancostanich, 2020-06-05:
Thanks for confirming the fix!

closed 2020-06-05 by bryancostanich