WildernessLabs/Meadow_Issues #88: Board work only if plugged in USB port

rudizz, 2020-04-01:
I deployed the Hello, Meadow sample in the board and everything is good, I can see the led pulse in different colors.
But if I power the board through a 5V plug (Samsung's charger) or a LiPo battery, I never see the led on, it seems that the board is not starting up.
I have measured the voltage across several pins and have always found that it is the same as when I connect the card via the USB port on my laptop. But with the 5V of the wall socket it does not work, with the 5V of the USB socket of the laptop it does. I can't understand why.
Is there any difference?
Am I missing something?
[image]
[image]

bryancostanich, 2020-04-02:
Sorry about that, this was a regression in `b3.8`.

We've got it fixed in `b3.10`, which we're in the process of testing/merging/validating.

rudizz, 2020-04-02:
ok, thank you for the quick reply.

closed 2020-04-04 by bryancostanich

rudizz, 2020-04-04:
Now it works.
Thanks