WildernessLabs/Meadow_Issues #93: Rider support

tomesendam, 2020-06-27:
I'm not sure if this is the right place but: I would like to see an extension for JetBrains rider. 
Are there plans for that?

bryancostanich, 2020-06-27:
No plans, but all the deployment tooling is open source, if you'd like to create one:

* [Meadow.CLI](https://github.com/wildernesslabs/Meadow.CLI) - Command Line Interface and core deployment libraries.
* [Visual Studio for Windows Extension](https://github.com/WildernessLabs/VS_Win_Meadow_Extension) - Code for the VSWin extension.
* [Visual Studio for Mac Extension](https://github.com/WildernessLabs/VS_Mac_Meadow_Extension) - Code for the VS4Mac extension.

closed 2020-06-27 by bryancostanich

tomesendam, 2020-06-27:
Well summer is right around the corner. Guess I'll try!