WildernessLabs/Meadow_Issues #94: Too many threads cause a crash

bryancostanich, 2020-06-27:
I've been doing some digging into the issue where Meadow crashes after a little while. I think there are two issues here. The first is a crash due to the Garbage Collector not operating correctly, which we have a fix for in our upcoming b4.0 release. The second, and still remaining issue seems to be around creating threads. I've whittled the repro down to a minimum app that does it. App is attached. 

What I suspect is happening is that thread handles (or something) aren't being cleaned up. One of the learnings is that `Task`s fail much faster than `Thread`s:

| duration | Task/Thread | Cycles | Uptime |
|----------|-------------|--------|--------|
| 1ms      | `Task`      | 770    | 16s    |
| 3ms      | `Task`      | 770    | 18s    |
| 10ms     | `Task`      | 770    | 24s    |
| 1ms      | `Thread`    | 15,040 | 2m:27s |
| 3ms      | `Thread`    | 15,040 | 2m:57s |
| 10ms     | `Thread`    | 15,040 | 4m:43s |

App:

[Meadow_Crasher_Diag.zip](https://github.com/WildernessLabs/Meadow_Issues/files/4841256/Meadow_Crasher_Diag.zip)

kangaroo, 2020-06-27:
@bryancostanich I'm on a new laptop, so setting things up is taking time.  Do you have the nuttx crash dump?  Any console output before it dies?

bryancostanich, 2020-07-05:
Yassssssss!!! Thanks @kangaroo!! Just tested and confirmed the fix. Thanks for the heroic fix.

closed 2020-07-15 by bryancostanich

brucecrombie, 2020-07-28:
Hey Bryan, before I go and post another issue for this, does "Fixed in Upcoming Release" mean b4.0 (yet to be released) or did you expect that the fix is in b3.12.  I seeing this behavior with b3.12. Much improved over previous releases, but still there.