WildernessLabs/Meadow_Issues #97: BME280 driver issue on V3.12

RobbyRobbyRobby, 2020-07-15:
Hi,

I've just upgraded to the new 3.12 release.  Sadly it doesn't work with the BME280.

It won't compile the old code (working under 3.11) bme280.StartUpdating().  

It complains as follows:

`reference to type 'FilterableObservableBase<,>' claims it is defined in 'Meadow.Foundation', but it could not be found.`

[image]

I suspect it's related to the FilterableObserver Changes made in this release.

Thanks,
Robin.

bryancostanich, 2020-07-17:
Hey Robin, did you get this working? I thought I saw something on the public slack that there was a mismatch on our publishing but with the new nugets it worked again.

bryancostanich, 2020-08-25:
Closing this issue, I can't repro, and I believe this was just a published nuget mismatch that got fixed.

closed 2020-08-25 by bryancostanich