WildernessLabs/Meadow_Issues #98: ISerialPort.Read does not remove bytes from the ReceiveBuffer anymore.

jjvenala, 2020-07-16:
Hi,

It seems that the ISerialPort.Read doesn't remove bytes from the circular receive buffer anymore, although the documentation states so and it used to do so. I can overcome this issue by calling ISerialPort.ClearReceiveBuffer after reading, for example.

I'm using 
- the latest FW (beta 3.12)
- Meadow.Foundation V0.17.0
- Meadow V0.15.0
- Meadow.Core V0.3.225

lionel-loscos, 2020-07-25:
I can confirm this issue.
A simple workaround is to use ISerialPort.ClearReceiveBuffer after a ISerialPort.Read.

svedmark, 2020-07-31:
Another piece of info: I haven't had this problem using ReadByte() .

bryancostanich, 2020-08-25:
@ctacke looks like this got introduced in your updates to the CircularBuffer. [This commit](https://github.com/WildernessLabs/Meadow.Core/commit/4501c557489528d0424f1a7e6639605e881f92dd#diff-1c65ce225bf96dc8ae2cea06aa907328). 

`Read()` used to call `Remove()` on the `_readBuffer`, now it only calls `MoveItemsTo()`.

closed 2020-09-22 by bryancostanich