WildernessLabs/Meadow_Issues #99: Unable to use System.Numerics.Vectors

craigktreasure, 2020-07-20:
I'm unable to configure an app to use types from `System.Numerics.Vectors` and have it run successfully. I have tried adding the `System.Numerics.Vectors` NuGet package directly, but the result is the same.

An exception is thrown:
`System.IO.FileNotFoundException, Could not load file or assembly 'System.Numerics, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089' or one of its dependencies.`

**Meadow Firmware**: 0.3.12
**VS Extension**:          0.10.1

MarkAlanJones, 2020-08-03:
I saw this issue as well

I believe it is related to the nuget having .net standard and .net 4.7.2 sets of dlls -
we compile to .net 4.7.2 - but the runtime is .net standard.
at deployment the nuget does not have the core libraries 

I tried forcing some of the other libraries to deploy, but it didn't help due to compiling to 4.7.2

adrianstevens, 2020-08-15:
We're not shipping the entire BCL set today so System.Numerics.dll isn't available. @bryancostanich @lamebrain I wonder if we could/should provide the other BCL dlls as an optional download with B4.0?

adrianstevens, 2020-10-29:
This is now fixed in B4.0.1 - thanks again Craig & Mark - closing issue

closed 2020-10-29 by adrianstevens