WildernessLabs/Meadow_Issues #105: Unable to flash the latest OS version to Meadow device

JimiSmith, 2020-09-22:
I'm using the latest Visual Studio extension (I updated immediately prior to opening the issue). Updating both of my Meadow devices to the latest OS fails in the same way. This is the log I get from Visual Studio

[2020/09/22 10:52:57] Begin 'Flash Device'
[2020/09/22 10:52:57] Upload Meadow.OS.bin (~2 mins)
[2020/09/22 10:54:46] Initialize device
[2020/09/22 10:54:49] Erase flash (~3 mins)
[2020/09/22 10:59:49] An unexpected error occurred. Please try again.

JimiSmith, 2020-09-22:
I was able to resolve this by erasing the flash using Meadow CLI. Erasing flash from the visual studio resulted in the same unexpected error. Using Meadow CLI first I disabled mono ( i had to run that 3 or 4 times before it worked) then I could erase flash. Once I'd done that then updating the OS from Visual Studio worked

closed 2020-09-23 by JimiSmith

bryancostanich, 2020-10-29:
fixed in b4.0.1