WildernessLabs/Meadow_Issues #106: Full Redeploy on small code change

AndrewFraser-Tendeka, 2020-09-22:
As mentioned on the forum, I am finding that a full deploy is performed every time a small code change is made. This results in me having to wait 20 minutes for each change to be deployed.

1>------ Build started: Project: 24AA02E48, Configuration: Debug Any CPU ------
2>------ Deploy started: Project: 24AA02E48, Configuration: Debug Any CPU ------
2>[20/09/20 12:32:23] Deploying to Meadow on COM7…
2>[20/09/20 12:32:23] Initializing Meadow (00:00:00.0000658 since last.)
2>[20/09/20 12:32:29] Device (00:00:06.3153786 since last.)
2>[20/09/20 12:32:29] Checking files on device (may take several seconds) (00:00:00.0223750 since last.)
2>[20/09/20 12:32:59] Deploying for the first time may take several minutes. (00:00:30.3191172 since last.)
2>[20/09/20 12:32:59] Writing App.exe (00:00:00.0362130 since last.)
2>[20/09/20 12:36:19] Writing Meadow.dll (**00:03:20**.0514533 since last.)
2>[20/09/20 12:39:39] Writing Meadow.Foundation.dll (**00:03:20**.0169533 since last.)
2>[20/09/20 12:42:59] Writing mscorlib.dll (**00:03:20**.0454271 since last.)
2>[20/09/20 12:46:19] Writing System.Core.dll (**00:03:20**.0084705 since last.)
2>[20/09/20 12:49:39] Writing System.dll (**00:03:19.999**7531 since last.)
2>[20/09/20 12:52:59] Resetting Meadow and starting app (30-60s) (00:03:20.0023429 since last.)
2>[20/09/20 12:53:02] Deployment Duration: 00:20:39.3371541 (00:00:02.5190878 since last.)
========== Build: 1 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
========== Deploy: 1 succeeded, 0 failed, 0 skipped ==========

Deploy started at 12:32 and took 20:39.833 minutes

It is interesting that several DLLs take more or less the same time to deploy 00:03:20 - could this be a timeout ?

svedmark, 2020-09-22:
I had a very similar situation when forgetting to update all parts - firmware/vs-extension/nugets (don't remember which I'd missed) - to the latest version.  On the line starting with "Device" you should see a version number if everything is all right.  Something like this:

1>[2020-09-22 19:59:31]     Device  MeadowOS Version: 0.4.0 (Sep 19 2020 14:37:21)                                                                   (00:00:09.6055673 since last.)

FraserElectronics, 2020-09-22:
I am pretty sure everything is up to date but even if it is not, surely the extension checks the version of the assemblies on the Meadow board and does not try to download them if they are already there ? 
This *normally* works where it just downloads the app and only takes a few seconds.

Andy

adrianstevens, 2020-10-29:
The Meadow extension does check the files on device before deploying. @AndrewFraser-Tendeka, are you still seeing this issue with the latest update? (B4.0.1)

bryancostanich, 2020-12-05:
@adrianstevens i think we can close this now. but let me know your thoughts.

AndrewFraser-Tendeka, 2020-12-08:
I have been able to find time to run Meadows with latest firmware and am not seeing the slow deployment issues.
Hopefully this issue is now resolved.

Best Regards

closed 2020-12-12 by bryancostanich