WildernessLabs/Meadow_Issues #107: Unable to resolve subdomain's

r-proto, 2020-09-24:
When calling a subdomain, the httpclient always errors with the following message "No such host is known".

NevynUK, 2021-07-06:
I have attempted to connect to a subdomain using the following code:

```c#
HttpResponseMessage response = await client.GetAsync("http://developer.wildernesslabs.co/");
response.EnsureSuccessStatusCode();
string responseBody = await response.Content.ReadAsStringAsync();
responseBody = await response.Content.ReadAsStringAsync();
Console.WriteLine(responseBody);
```

I receive a good response from the server.

Can you please provide an example for me to try.

r-proto, 2021-07-07:
@NevynUK I haven't tried this in any of the new versions, so the issue may be resolved.  I will try later this week.

bryancostanich, 2021-08-08:
no update for 30days, closing. @r-proto if this is still an issue, please re-open.

closed 2021-08-08 by bryancostanich