WildernessLabs/Meadow_Issues #108: System.IO.FileNotFoundException: Could not load file or assembly 'netstandard, Version=2.0.0.0

r-proto, 2020-09-24:
I have a netstandard 2.0.3 class library I am referencing and as soon as I call any code in that library it throws:

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'netstandard, Version=2.0.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51' or one of its dependencies.
File name: 'netstandard, Version=2.0.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51'
  at MeadowApp.MeadowApp..ctor () <0xc10130e8 + 0x0000a> in <d940124cdcea40dba21eaff2b7486c39>:0

  at MeadowApp.Program.Main (System.String[] args) <0xc1012f40 + 0x00000> in <d940124cdcea40dba21eaff2b7486c39>:0 

To verify that this wasn't a problem with any other references I created a sample project that only has string extensions in the netstandard library and I get the same error.

adrianstevens, 2020-11-18:
.NET Standard 1.0 through 1.6 should work. We still need to investigate .NET Standard 2.0.

bryancostanich, 2021-05-10:
This should be fixed in the upcoming b5.0 release.

closed 2021-05-25 by bryancostanich