WildernessLabs/Meadow_Issues #113: System.Collections.Generic.KeyNotFoundException when creating a Digital Input Pin or BiDirectional Pin

mjjames, 2020-10-04:
If I create a Digital Input Port
``` c#
try
{
    using (var port = Device.CreateDigitalInputPort(pin))
    {
        Console.WriteLine($"Port Created for: {pin.Name} |{pin} ");
    }
}
catch (Exception ex)
{
    Console.WriteLine($"Create Port Failed for: {pin.Name} | {pin}");
    Console.WriteLine(ex);
}
```
I get a KeyNotFoundException withing WireInterrupt.
``` c#
System.Collections.Generic.KeyNotFoundException: The given key '9' was not present in the dictionary.
  at System.Collections.Generic.Dictionary`2[TKey,TValue].get_Item (TKey key) <0xc12a9970 + 0x00038> in <710bb3a08ed44255b7414b4b722d99f2>:0 
  at Meadow.Devices.F7GPIOManager.WireInterrupt (Meadow.Core.Interop+STM32+GpioPort port, System.Int32 pin, Meadow.Hardware.InterruptMode i terruptMode, Meadow.Core.Interop+STM32+ResistorMode resistorMode, System.Double debounceDuration, System.Double glitchDuration) <0xc12a2cb0 + 0x002a4> in <cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at Meadow.Devices.F7GPIOManager.ConfigureGpio (Meadow.Core.Interop+STM32+GpioPort port, System.Int32 pin, Meadow.Core.Interop+STM32+GpioMode mode, Meadow.Core.Interop+STM32+ResistorMode res stor, Meadow.Core.Interop+STM32+GPIOSpeed speed, Meadow.Core.Interop+STM32+OutputType type, System.Boolean initialState, Meadow.Hardware.InterruptMode interruptMode, System.Int32 alternateFunctionNumber, System.Double debounceDuration, System.Double glitchDuration) <0xc113afa8 + 0x0036e> in <cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at Meadow.Devices.F7GPIOManager.ConfigureGpio (Mead w.Hardware.IPin pin, Meadow.Core.Interop+STM32+GpioMode mode, Meadow.Core.Interop+STM32+ResistorMode resistor, Meadow.Core.Interop+STM32+GPIOSpeed speed, Meadow.Core.Interop+STM32+OutputType type, System.Boolean initialState, Meadow.Hardware.InterruptMode interruptMode, System.Double debounceDuration, System.Double glitchDuration) <0xc1299720 + 0x0006a> in <cae2d84da7b047bf9a8c690 4ebf763e>:0 
  at Meadow.Devices.F7GPIOManager.ConfigureInput (Meadow.Hardware.IPin pin, Meadow.Core.Interop+STM32+ResistorMode resistor, Meadow.Hardware.InterruptMode interruptMode, System.Double debounceDuration, System.Double glitchDuration) <0xc1298f48 + 0x000f0> in <cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at Meadow.Devices.F7GPIOManager.ConfigureInput (Meadow.Hardware.IPin pin  Meadow.Hardware.ResistorMode resistorMode, Meadow.Hardware.InterruptMode interruptMode, System.Double debounceDuration, System.Double glitchDuration) <0xc128fee8 + 0x00044> in <cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at Meadow.Hardware.DigitalInputPort..ctor (Meadow.Hardware.IPin pin, Meadow.Hardware.IIOController ioController, Meadow.Hardware.IDigitalChannelInfo channel, Meadow.H rdware.InterruptMode interruptMode, Meadow.Hardware.ResistorMode resistorMode, System.Double debounceDuration, System.Double glitchDuration) <0xc12621e0 + 0x000c8> in <cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at Meadow.Hardware.DigitalInputPort.From (Meadow.Hardware.IPin pin, Meadow.Hardware.IIOController ioController, Meadow.Hardware.InterruptMode interruptMode, Meadow.Hardware.Res storMode resistorMode, System.Double debounceDuration, System.Double glitchDuration) <0xc125a328 + 0x000e2> in <cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at Meadow.Devices.F7Micro.CreateDigitalInputPort (Meadow.Hardware.IPin pin, Meadow.Hardware.InterruptMode interruptMode, Meadow.Hardware.ResistorMode resistorMode, System.Double debounceDuration, System.Double glitchDuration) <0xc12 a078 + 0x0001a> in <cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at TempMonitor.MeadowApp+<>c.<TestPinAsInput>b__3_0 (Meadow.Hardware.IPin p) <0xc1259f98 + 0x00026> in <43ef7be0a8744700af098b76bee83761>:0 
  at (wrapper delegate-invoke) System.Func`2[Meadow.Hardware.IPin,Meadow.Hardware.IDigitalPort].invoke_TResult_T(Meadow.Hardware.IPin)
  at TempMonitor.MeadowApp.TryCreatePort (Meadow Hardware.IPin pin, System.Func`2[T,TResult] createPort) <0xc1259818 + 0x00012> in <43ef7be0a8744700af098b76bee83761>:0 
Interupt Edge Both
Create Port Failed for: D00 | D00
 Meadow.Hardware.PortInUseException: Exception of type 'Meadow.Hardware.PortInUseException' was thrown.
  at Meadow.Hardware.DigitalInputPort..ctor (Meadow.Hardware.IPin pin, Meadow.Hardware.IIOController ioController, Meadow.Hardware.IDigitalChannelInfo channel, Meadow.Hardware.InterruptMode interruptMode, Meadow.Hardware.ResistorMode resistorMode, System.Double debounceDuration,  ystem.Double glitchDuration) <0xc12621e0 + 0x000d8> in <cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at Meadow.Hardware.DigitalInputPort.From (Meadow.Hardware.IPin pin, Meadow.Hardware.IIOController ioController, Meadow.Hardware.InterruptMode interruptMode, Meadow.Hardware.ResistorMode resistorMode, System.Double debounceDuration, System.Double glitchDuration) <0xc125a328 + 0x000e2> in  cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at Meadow.Devices.F7Micro.CreateDigitalInputPort (Meadow.Hardware.IPin pin, Meadow.Hardware.InterruptMode interruptMode, Meadow.Hardware.ResistorMode resistorMode, System.Double debounceDuration, System.Double glitchDuration) <0xc125a078 + 0x0001a> in <cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at TempMonitor.MeadowApp+<>c.<TestPinAsInput>b__3_1  Meadow.Hardware.IPin p) <0xc12f25e8 + 0x00026> in <43ef7be0a8744700af098b76bee83761>:0 
  at (wrapper delegate-invoke) System.Func`2[Meadow.Hardware.IPin,Meadow.Hardware.IDigitalPort].invoke_TResult_T(Meadow.Hardware.IPin)
  at TempMonitor.MeadowApp.TryCreatePort (Meadow.Hardware.IPin pin, System.Func`2[T,TResult] createPort) <0xc1259818 + 0x00012> in <43ef7be0a8744700af098b76bee8 761>:0 
```

This also causes a second fault, if you then try to create an input again immediately after that code for the same pin but different setup i.e set Interupt Both Edges you get a port in use exception. So it's created the port, failed to wire up the interrupt and then doesn't dispose the port to allow it to be used again.

``` c#
Meadow.Hardware.PortInUseException: Exception of type 'Meadow.Hardware.PortInUseException' was thrown.
  at Meadow.Hardware.DigitalInputPort..ctor (Meadow.Hardware.IPin pin, Meadow.Hardware.IIOController ioController, Meadow.Hardware.IDigitalChannelInfo channel, Meadow.Hardware.InterruptMode interruptMode, Meadow.Hardware.ResistorMode resistorMode, System.Double debounceDuration,  ystem.Double glitchDuration) <0xc12621e0 + 0x000d8> in <cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at Meadow.Hardware.DigitalInputPort.From (Meadow.Hardware.IPin pin, Meadow.Hardware.IIOController ioController, Meadow.Hardware.InterruptMode interruptMode, Meadow.Hardware.ResistorMode resistorMode, System.Double debounceDuration, System.Double glitchDuration) <0xc125a328 + 0x000e2> in  cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at Meadow.Devices.F7Micro.CreateDigitalInputPort (Meadow.Hardware.IPin pin, Meadow.Hardware.InterruptMode interruptMode, Meadow.Hardware.ResistorMode resistorMode, System.Double debounceDuration, System.Double glitchDuration) <0xc125a078 + 0x0001a> in <cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at TempMonitor.MeadowApp+<>c.<TestPinAsInput>b__3_1  Meadow.Hardware.IPin p) <0xc12f25e8 + 0x00026> in <43ef7be0a8744700af098b76bee83761>:0 
  at (wrapper delegate-invoke) System.Func`2[Meadow.Hardware.IPin,Meadow.Hardware.IDigitalPort].invoke_TResult_T(Meadow.Hardware.IPin)
  at TempMonitor.MeadowApp.TryCreatePort (Meadow.Hardware.IPin pin, System.Func`2[T,TResult] createPort) <0xc1259818 + 0x00012> in <43ef7be0a8744700af098b76bee8 761>:0 
```

The KeyNotFound exception also occurs for BiDirectionalDigitalPorts within the Wireup Interupts method:
``` c#
System.Collections.Generic.KeyNotFoundException: The given key '9' was not present in the dictionary.
  at System.Collections.Generic.Dictionary`2[TKey,TValue].get_Item (TKey key) <0xc12a9160 + 0x00038> in <710bb3a08ed44255b7414b4b722d99f2>:0 
  at Meadow.Devices.F7GPIOManager.WireInterrupt (Meadow.Core.Interop+STM32+GpioPort port, System.Int32 pin, Meadow.Hardware.InterruptMode i terruptMode, Meadow.Core.Interop+STM32+ResistorMode resistorMode, System.Double debounceDuration, System.Double glitchDuration) <0xc12a2c70 + 0x002a4> in <cae2d84da7b047bf9a8c690a4ebf763e>:0 
  at Meadow.Devices.F7GPIOManager.ConfigureGpio (Meadow.Core.Interop+STM32+GpioPort port, System.Int32 pin, Meadow.Core.Interop+STM32+GpioMode mode, Meadow.Core.Interop+STM32+ResistorMode res stor, Meadow.Core.Interop+STM32+GPIOSpeed speed, Meadow.Core.Interop+STM32+OutputType type, System.Boolean initialState, Meadow.Hardware.InterruptMode interruptMode, System.Int32 alternateFunc [... cut ...]
  at TempMonitor.MeadowApp+<>c.<TestPinAsBiDirectional>b__4_0 (Meadow.Hardware.IPin p) <0xc1259aa0 + 0x0002c> in <d0944805844d41599ee1ce6b1789c7bb>:0 
  at (wrapper delegate-invoke) System.Func`2[Meadow.Hardware.IPin, Meadow.Hardware.IDigitalPort].invoke_TResult_T(Meadow.Hardware.IPin)
  at TempMonitor.MeadowApp.TryCreatePort (Meadow.Hardware.IPin pin, System.Func`2[T,TResult] createPort) <0xc1278340 + 0x00012> in <d0944805844d41599ee1ce6b1789c7bb>:0 
```

Device  MeadowOS Version: 0.4.0 (Sep 19 2020 14:37:21)     
Meadow.Foundation: 0.18.0

adrianstevens, 2020-10-29:
@mjjames - thank you, I was able to reproduce the issue. We'll get this fixed in the next release.

@ctacke - it looks like Meadow is throwing KeyNotFoundException when the InterruptMode is set to None (which is also the default)

adrianstevens, 2021-08-09:
@mjjames @bryancostanich @ctacke this is fixed in B5.2 (was probably fixed in 5.0) 

Closing issue

closed 2021-08-09 by adrianstevens