WildernessLabs/Meadow_Issues #116: System.Net.Sockets.SocketException (0x80004005): Access denied on creating Socket

SebaWnek, 2020-10-09:
When trying to create System.Net.Sockets.Socket either through Socket or UdpClient or TcpListener it throws  
`Unhandled Exception: System.Net.Sockets.SocketException (0x80004005): Access denied
  at WiFi_Basics.MeadowApp..ctor () <0xc1013dc0 + 0x00096> in <eabb90eab81d4b2fb67b644dd4dc2be3>:0 
  at WiFi_test.Program.Main (System.String[] args) <0xc10137c8 + 0x00034> in <eabb90eab81d4b2fb67b644dd4dc2be3>:0 `

Example code [https://github.com/SebaWnek/WiFi-test/blob/master/WiFi%20test/MeadowApp.cs] - here I've tried with Socket in UDP mode and TCP through TcpListener, before I've tried UdpClient and result was exactly the same (didn't bothered to finish TCP code as it's unable to open Socket anyway)

richardjharding, 2020-10-12:
I was able to replicate this with Sebastians sample code
this function:
```
private void ConnectToClientTCP()
        {
            ipLocalEP = new IPEndPoint(IPAddress.Parse("192.168.148.5"), 22222);
            ipClientEP = new IPEndPoint(IPAddress.Parse("192.168.44.97"), 33333);
            TcpListener listener = new TcpListener(ipLocalEP);
            try
            {
                Console.WriteLine("Starting tcp listener");
                listener.Start();
                Console.WriteLine("Connected!");
                onboardLed.SetColor(Color.Blue);
                //display.Write("Ready!");
            }
            catch (Exception e)
            {
                //display.Write(e.Message);
                throw;
            }
        }
```
throws on ```listener.Start()``` 

the meadow was connected to my wifi with an IP of 192.168.148.5 ie the ipLocalEP variable

this is all with beta4 deployed

NevynUK, 2020-12-05:
UdpClient, TcpListener and TcpClient tested and working ready for the next release.

SebaWnek, 2020-12-05:
great news! Can't wait :) hopefully I'll be done with mechanical parts finally by then :)

NevynUK, 2020-12-12:
Issue was fixed in beta 4.2 released 10 Dec.

closed 2020-12-12 by NevynUK