WildernessLabs/Meadow_Issues #120: Unable to deploy F# application

bjorndaniel, 2020-12-20:
I seem to be unable to deploy any F# applications to the Meadow using Visual Studio 2019.
If I create a new C# application it deploys right away but doing the same with F# yields the following error:
Could not find file 'C:\...\bin\Debug\net472\netstandard.dll'.
My environment is:
Windows 10 
VS 2019 16.8.3
Meadow extension 0.13.0
Meadow.Foundation 0.20.0
FSharp.Core 5.0.0
Any ideas?

mongeon, 2020-12-21:
Looks like the same #108

bjorndaniel, 2020-12-21:
#108 seems to be a runtime error which to me indicates that the code has been deployed?
In my case it never even pushes the files over before giving the deployment failed error in the output window of VS.

adrianstevens, 2020-12-31:
The behaviour of the extension changed in the last release so it's failing at deploy-time now for missing dlls. It may get changed back in the next release.

For this issue specifically, it looks like the dependancies for F# 5 have changed which is why this recently broke.

We either need to fix the NetStandard.dll issue - which we may not be able to fix until the linker is working, or force an earlier version of F# (not certain if that's possible).

adrianstevens, 2020-12-31:
I put some time into this but haven't had any luck. NetStandard.dll is needed for F# 5.0 so omitting the dll isn't an option. Adding NetStandard.dll is a bit of a reference mess, which wouldn't normally be an issue but Meadow doesn't have linker support yet.

You can force earlier versions of F# by setting the language version in the project but it's still using the F# 5.0 SDK Nuget.

It's also possible to reference earlier F# nuget versions but since the SDK is pulling in 5.0 that will take priority over earlier versions.

bjorndaniel, 2021-01-01:
Thanks for checking, I guess I'll have to stick with C# for now then.

aonyxrocks, 2021-01-17:
+1. I just ran into the same issue.

bryancostanich, 2021-05-10:
@adrianstevens we should test with b5.0, this should be fixed now.

bryancostanich, 2021-05-14:
This is fixed in b5.0 with the inclusion of the required dlls into the MeadowAssemblies nupkg, and the linker.

closed 2021-05-25 by bryancostanich