WildernessLabs/Meadow_Issues #123: WiFiAdapter incorrectly reporting successful connection

devingoble, 2021-01-01:
### My board
`ID: Meadow by Wilderness Labs, Model: F7Micro, MeadowOS Version: 0.4.2 (Dec  8 2020 10:09:54), Processor: STM32F777IIK6, Processor Id: 21-00-1b-00-07-51-38-31-31-39-35-35,Serial Number: 3550395C3138, CoProcessor: ESP32, CoProcessor OS Version: 0.4.2.0`

### Issue
Given the following code, where the SSID and Key are deliberately incorrect values:
```
Device.InitWiFiAdapter().Wait();

var ssid = "asdf";
var key = "asdfasdf";

var status = Device.WiFiAdapter.Connect(ssid, key).ConnectionStatus;

if(status == ConnectionStatus.Success)
{
    Console.WriteLine("Connected to " + ssid);
    Console.WriteLine("Has Internet: " + Device.WiFiAdapter.HasInternetAccess);
}
else
{
    Console.WriteLine("Cannot connect");
    Console.WriteLine(status.ToString());
}
```

`status` is always `ConnectionStatus.Success`.

Am I correctly understanding how to use this API, or is this a bug?

devingoble, 2021-01-05:
I updated everything to b4.3 last night and I'm still getting the same result.

joreg, 2021-01-25:
same here! i used to have an acutally successful connection though before updating to b4.4!

Simas1, 2021-01-26:
Hi, I have the same issues with Wifi: ConnectionStatus.Success, Device.WiFiAdapter.HasInternetAccess, Device.WiFiAdapter.IsConnected - they all report incorrect status. I pass wrong SID, or I pass good SID but wrong password - these properties always return TRUE.

BeeKaiGee, 2021-01-31:
Hi, same issue here. Device.WiFiAdapter.Connect() always return ConnectionStatus.Success.

closed 2021-02-08 by NevynUK